fabianoriccardi/dimmable-light #1: Where the unordered map comes from?

MainMa, 2019-01-01:
Hello,

You've [used `unordered_map` in your library](https://github.com/fabiuz7/dimmable_light_esp8266_lib/blob/cb3fb3cfef1a97e24bc24fa54a8d660d48536d31/src/dimmable_light_manager.h#L23), which means that compiling it necessarily leads to an error like this:

> /home/arsene/Arduino/libraries/Dimmable_Light_for_ESP/src/dimmable_light_manager.h:32:25: fatal error: unordered_map: No such file or directory
>  #include <unordered_map>
>                          ^
> compilation terminated.
> exit status 1
> Error compiling for board Arduino/Genuino Uno.

Could you please document which third-party library you used to get the functionality of unordered map? It doesn't seem to be neither _ArduinoSTL_ by Mike Matera, nor _Embedded Template Library_ by John Wellbelove, so which one should be included?

fabiuz7, 2019-01-01:
You’re right, I forgot to mention that example #4 works only for esp8266! Is it easy the integration with these libraries?

MainMa, 2019-01-01:
It's not about the fourth example, but all of them. The steps to reproduce the issue are rather simple:

1. Within Arduino IDE, install _Dimmable Light for ESP_.
1. In _File_ › _Examples_, chose any of the examples provided in this library.
1. Compile.

I suppose that on your machine, this will compile, because you installed a third-party library in order to have partial STL support. But on any fresh installation, the error I quoted in my original message would appear.

fabiuz7, 2019-01-01:
I've just pushed the commit that should fix the problem. Now it requires ArduinoSTL. I've trying to be as minimalistinc as possible with this fix: I decided to use a _map_ instead of _unordered_map_, for <10 elements the performance differences should be irrelevant. ESP* continue to make use of unordered_map.

The next release (coming in few days) contains this fix.

Thanks for the feedback!

closed 2019-01-01 by MainMa