fabianoriccardi/dimmable-light #2: LED Flickering (9600rate)

dinomaskot, 2019-01-21:
Hi,

I've just tried your example with one light using robotdyn AC dimmer and esp8266 and I'm getting a lot of flickering.

Could it be because my board baud rate is 9600 instead of 115200?

Thank you.

mazeemsk, 2019-02-24:
Hi,  I have the same problem "using robotdyn AC dimmer and NodeMCU and I'm getting a lot of flickering." Frequency is 50HZ. Please Help me.

mazeemsk, 2019-03-20:
Hi, my problem is  I'm getting a lot of flickering. My frequency is 50HZ. Why is there flickering and how can I get rid of that.

fabiuz7, 2019-03-20:
First a couple of question: what is the circuit? Which voltage do you apply for Signal and Vcc pins? Do you always get flickering or only for specific brightness values?

mazeemsk, 2019-03-21:
1- using this circuit  http://arduinotehniq.blogspot.com/2014/10/ac-light-dimmer-with-arduino.html with nodemcu..... D6 for zero cross and D7 for signal....VCC for the circuit is 5V feed from nodemcu..
2-I always get flickering for all values....

fabiuz7, 2019-03-21:
Could you try with Vcc=3,3V? However that's weird, I'm planning to insert some lines of code to better understand your problem.

mazeemsk, 2019-03-21:
VCC is as per circuit design ... so in my opinion it should not be the problem.

RudyFiero, 2019-04-09:
@dinomaskot  Can you try it with an incandescent lamp in parallel with the led light? Triac dimmers do not always work properly with led lights. Often there is flicker in some of the range.

Mri007, 2019-06-12:
Zc pin GP4
Triac pin GP16
50hz freq
flicker noticable with incandescent lamp while dimming level less than 200.

fabiuz7, 2019-11-06:
Sorry for the long delay, but in that period I was really busy. Recently, I get some time to look back and improve at this library.

Had you finally solved the problem?

Mri007, 2019-11-06:
Ohh thanks for reinvesting your time. I didnt manage to solve the flickering issue no matter how hard i bang my head on the wall >.<
Any help will appreciate.

Karshilistic, 2019-11-06:
I would suggest the following sanity checks

1. Check that you're using the latest release of the library. 1.2.0 and earlier were causing flickering for me
2. Check different light bulbs, including incandescent and dimmable LEDs. I found that some light bulbs behave differently and could flicker, when others do not.
3. Try changing`static const uint16_t startMargin = 200;` and `static const uint16_t endMargin = 500;` in Thyristor.cpp. I played around these values in an earlier version and that seemed to help a little.
4. Check issue #12 and @fabiuz7's response there. They seem to be using a similar board to yours.

Good luck.

fabianoriccardi, 2023-11-09:
@Karshilistic thanks for the detailed troubleshooting, but after many years without any activity and many versions later, I'm closing this issue.

closed 2023-11-09 by fabianoriccardi