fabianoriccardi/dimmable-light #3: rete AC 60Hz

ivale, 2019-02-01:
Ho provato ad utilizzare la libreria su ESP32 con questo modulo 
https://www.amazon.it/gp/product/B07BK6899B/ref=ppx_yo_dt_b_asin_title_o05__o00_s00?ie=UTF8&psc=1 collegato alla rete AC italiana ma qualunque carico collego lampeggia. La libreria funziona con la AC? Grazie e buon lavoro

fabiuz7, 2019-02-01:
Strano, è progettato per funzionare in AC a 60Hz. Ho ricevuto pochi giorni fa lo stesso componente che mi hai linkato. Appena possibile lo provo e ti faccio sapere!

Grazie per il feedback

ivale, 2019-02-08:
Scusa per averti risposto in ritardo. Ho provato come prima cosa a passare ad un Arduino Mega e il flickering é scomparso ma mi si é proposto un nuovo problema. Prendendo il primo esempio  dimmable_light per i livelli di brightness superiori ai 215 circa, il modulo smette di erogare potenza ed inoltre il led sul modulo non raggiunge mai luminosità massima ad indicare che il pwmPin non eroga mai 5v. Se infatti lo collego diretto ai 5v il modulo funziona come dovrebbe. Come utilizzatore ho attaccato una classica luce a incandescenza a filamento. Non dovrebbe nemmeno essere un problema di codice dato che utilizzo il tuo esempio su Arduino mega originale. I pin che utilizzo sono syncPin=3 per lo zero-cross e thyristorPin=2 per il PWM. Nei prossimi giorni proverò a sostituire il modulo, fosse quello il problema.

Su ESP32 invece il problema del flickering rimane anche con la modifica che mi hai suggerito al file thyristor.cpp. Uso un modulo DOIT ESP32, pin D4 per il PWM e D23 per lo zero-cross.

Sbaglio qualcosa? I pin che utilizzo sono giusti?

Grazie mille!

fabiuz7, 2019-02-10:
Nei prossimi giorni proverò a controllare meglio, per ora potresti dirmi a che tensione alimenti il dimmer quando usi l'ESP32?

ivale, 2019-02-18:
ho provato ad alimentare sia con 3.3 che con 5v ma non cambia nulla. Ho provato anche ad usare diversi pin dell'ESP ma rimane il problema. Proverò ad usare un BJT per il PWM perché penso che il modulo assorbi tanto dato che anche con l'arduino (dove non ho il problema del flickering) non riesce mai a raggiungere la potenza massima.

charles, 2019-02-21:
Hi, do you have plans to release 60Hz support? Thanks.

fabiuz7, 2019-02-21:
Yes I'm going to support it. It should not require much effort, but of course I can't test it. Would you help me?

charles, 2019-02-22:
> Yes I'm going to support it. It should not require much effort, but of course I can't test it. Would you help me?

Of course, in Brazil the eletrical network is 60Hz, but I do not have much knowledge and electronic instrumentations like a oscilloscope, however, I can test your implementations on some lamps and check if it is not blinking or something weird when dimm the lamp.

fabiuz7, 2019-02-22:
Ok that's fine, may I ask you which dimmer module you are using? A photo/link is enough

charles, 2019-02-22:
I'm use this dimmer: https://www.usinainfo.com.br/dimmer-arduino/modulo-dimmer-para-arduino-pic-mc-8a-com-sinal-zero-cross-2799.html 
I test with your lib, and works ok with incandescent lamp and a dimmerable led spot, both 60Hz. The weird thing is that the lamp turns on in 30% of dimm and turns off if over them 94% of power, may be this happens because of frequency 50Hz set inside de lib. The power when 94% is around 120V, that is the correct current on eletrical network, however after 94% the lamp turns off. I don't know why.

fabiuz7, 2019-03-20:
Sorry if too much time has passed, please try latest commit. The proper define (NETWORK_FREQ_60HZ) is in file thyristor.h

fabiuz7, 2019-10-27:
> I'm use this dimmer: https://www.usinainfo.com.br/dimmer-arduino/modulo-dimmer-para-arduino-pic-mc-8a-com-sinal-zero-cross-2799.html
> I test with your lib, and works ok with incandescent lamp and a [...]

Even this problems (60Hz frequency and suddently turning off when over 94%) are solved in version 1.3.0

closed 2019-10-27 by fabiuz7