fabianoriccardi/dimmable-light #4: No interrupts with ESP32

RudyFiero, 2019-04-09:
I am using 1_dimmable_light and I am not getting any response from the output. It seems that interrupts are not working. I have tried other code to verify that the interrupts work on the pin I am using (GPIO16).  My sync input is a positive pulse just over 1mS and the period is 8.3mS. (60Hz)

Can you tell me how I could narrow down the problem? 

I am a hardware designer and not a programmer. My programming skills are limited, but improving. :)

fabiuz7, 2019-04-09:
That's really weird, I've just tried again and it works fine (also using pin 16/14 as cite in your code).

Have you properly un(commented) these [2 lines](https://github.com/fabiuz7/Dimmable-Light-Arduino/blob/9054a98fe0e55109c8dca69ec55265e5cc0c5875/src/thyristor.h#L41) ? They are the selector for network frequency.

RudyFiero, 2019-04-09:
I loaded your library from the Library Manager in Arduino. It is different than the file you linked to. I guess you have made a change to the code and it has not yet available through the manager. I will copy the repository from Github when I get home.

RudyFiero, 2019-04-10:
I did some tests with the current code and found that the light turns on fully for the duration of only one of the values of setBrightness. All other values produce no output.  The following is what I have done.

I print out the value of setBrightness

```
void loop() {
  for (int i = 252; i < 256; i++) {
    light.setBrightness(i);
    Serial.print("setBrightness ");
    Serial.println(i);
    delay(period);
  }
}
```

I added a print of newDelay
```
dimmable_light.h
  	/**
   	 * Set the brightness, 0 to turn off the lamp
   	 */
  	void setBrightness(uint8_t bri){
      brightness=bri;
#ifdef NETWORK_FREQ_50HZ
      uint16_t newDelay=10000-(uint16_t)(((uint32_t)bri*10000)/255);
#elif defined(NETWORK_FREQ_60HZ)
      uint16_t newDelay=8333-(uint16_t)(((uint32_t)bri*8333)/255);
#endif
      thyristor.setDelay(newDelay);

    Serial.print("newDelay ");
    Serial.println(newDelay);
    };

```

The result in the serial monitor is this.

```
newDelay 66
setBrightness 253
newDelay 33
setBrightness 254
newDelay 0                     |  light on fully
setBrightness 255           |  light on fully
newDelay 99
setBrightness 252
newDelay 66
setBrightness 253
newDelay 33
setBrightness 254
newDelay 0
setBrightness 255
newDelay 99
setBrightness 252
newDelay 66
setBrightness 253
```

When get  
newDelay 0
setBrightness 255
The output goes on full. (high) for the duration of     delay(period); that I have set to 7000. At any other value other than this one above, the port pin is low and the triac is never triggered.

fabiuz7, 2019-04-10:
With the last update, setting brightness=255 means that the pin connected to the thyristor is always ON. If so, the sync interrupt is automatically disabled.

I'm writing a function for network frequency autodetection. When I'll get home I'll test it. This could help to understand why this interrupt do not trigger. (I've re-tested this code with Lolin D32 and it works finely)

RudyFiero, 2019-04-12:
I came across something that I think might be relevant to this issue. I have been using a revision 0 ESP32 to test this code. I have also been trying to get my own code to work but I have been getting inconsistent results. 

I found an issue on the ESP32-Arduino repository about not being able to enable, disable, then re-enable interrupts for the timer. But the problem was on a Rev 0 part, a Rev 1 did not have the same problem. So I think the problem I was having was because of an issue with the Rev 0 parts, and I think you probably have a Rev 1 part and that is why you could not see the same problem.

fabiuz7, 2019-04-12:
This seems a good point!
But what is rev 0 or rev 1? Hardware revision or software? About hardware I use Lolin D32 and about esp32-arduino i'm using v1.0.1.

RudyFiero, 2019-04-12:
Hardware revision. 

https://www.instructables.com/id/How-to-Check-the-Revision-Level-of-Your-ESP32/

fabiuz7, 2019-04-12:
Yeah it seems related... Also I've just checked my board

Chip is ESP32D0WDQ6 (revision 1)

Actually I don't know if I have a rev 0 to try to fix the bug...

RudyFiero, 2019-04-12:
Maybe it can't be fixed. 

I have been trying to do my own code to do phase control of a triac. Electric heat rather than lighting. Single output, simpler than what you have done here. Less critical. I don't have to worry about lamp flicker. I just need average control.

What I did should have worked, and I couldn't understand why it didn't. I am going to adapt a board with a newer revision of the chip, get my code working, then see if I can get the rev 0 to work. I will let you know how it goes and if I can find a workaround for this problem.

fabiuz7, 2019-09-29:
Hi @RudyFiero,
I have just pushed a fix that hopefully solves the problem of esp32 rev0. If you mind to test it, I would be glad to hear your feedback

RudyFiero, 2019-09-29:
I don't have a working setup at this time. I will try this but it might be a few days.

fabiuz7, 2019-10-27:
New version 1.3.0 supporting rev0 is out :)

closing as for #8

closed 2019-10-27 by fabiuz7