fabianoriccardi/dimmable-light #6: D7 not declared in this scope

waipeinan, 2019-06-12:
first attempt in arduinio with 1 dimmable light for an esp32, and getting both D5 and'D7' not declared in this scope errors?

Im assuming i'm making a schoolboy error in this yet would like a hand if possible.
ta
s

Mri007, 2019-06-13:
try using the gpio number instead of the silkscreen number.
https://circuits4you.com/wp-content/uploads/2018/12/ESP32-Pinout.jpg

waipeinan, 2019-06-15:
Thank you very much
that worked perfectly.

***
*Simon Peter Tyszko*
*www.theculture.net <http://www.theculture.net>*
*www.phlight.org <http://www.phlight.org>*
*Isotopica episodes on Resonance FM
<https://www.mixcloud.com/Resonance/playlists/isotopica/>*
*Isotopica <http://www.theculture.net/feed>*
*instagram <https://www.instagram.com/tyszko.isotopica/>*

closed 2019-11-06 by fabiuz7