fabianoriccardi/dimmable-light #7: AC 60Hz

robertowiest, 2019-07-16:
Hello, how can I use this library on a 220v (60Hz) voltage network with an ESP32?

fabiuz7, 2019-07-16:
Hello!
In which sense? You should be able to download the updated version from Arduino IDE

robertowiest, 2019-07-16:
Sorry, I mispronounced. To use the lib on a 200v (60Hz) network with an ESP32 module and its first example. Do I need to specify these voltage and module characteristics somewhere? Or just specify the ports const int syncPin = D7 and const int thyristorPin = D5 ;?

fabiuz7, 2019-07-16:
Yes, it should work without any other setup. This is a library to control directly digital pins, so it should require any other modification. By default, the library is set to 60Hz (the voltage value is not relevant, your module should convert between AC and 5V signal).

robertowiest, 2019-07-23:
In my pilot project I developed with arduino using a lamp, until then it was working perfectly. However, I am wanting to plug it into a 360W electrical resistor and would like to know if it is possible. Because I am turning on the lamp in parallel and it oscillates the brightness constantly.

When checking the voltage with a multimeter I found that it is oscillating between 80 and 220v. But with only the lamp the voltage is constant at 220v.

Any suggestions what it might be?

fabiuz7, 2019-10-31:
Hi Roberto,
did you solve the issue?

Karshilistic, 2019-11-06:
> Yes, it should work without any other setup. This is a library to control directly digital pins, so it should require any other modification. By default, the library is set to 60Hz (the voltage value is not relevant, your module should convert between AC and 5V signal).

I believe the default is actually 50Hz. In any case, comment and uncomment the relevant frequency in https://github.com/fabiuz7/Dimmable-Light-Arduino/blob/480a8a321623d385440a681a5eae509313875a25/src/thyristor.h#L41-L42

natelefever, 2020-11-29:
Switching it to 60Hz mode breaks some code in dimmable_light_linearized.h - line 62 is missing a ; 

Just FYI :)

fabianoriccardi, 2023-11-09:
> Switching it to 60Hz mode breaks some code in dimmable_light_linearized.h - line 62 is missing a ;

Solved.  

The lasted question of Robert was not answered, but after staling so many years I'm closing this issue. If still relevant, just ping this topic and I will reopen it.

closed 2023-11-09 by fabianoriccardi