fabianoriccardi/dimmable-light #8: no success on ESP32 doit devkit 1, 1_dimmable_light_example

TLS1000, 2019-09-01:
Hi, I'm using the esp32 doit V1, with sync pin 35, and output pin 25, library version 1.2.0.

I get a 50ms pulse on my output every ~12.5s. It's constant interval, not random.
Seems like some problem with timer scaler or so?

fabiuz7, 2019-09-01:
Hi TLS1000,
have you tried with different pins? May it be related to #4 ?

TLS1000, 2019-09-01:
Hi Fabiuz, I haven't tried that yet. But I found out that when I set the brightness to 255, the output goes on all the time. So the 12.5s cycle was just because of the loop with rising brightness reaching 255.

I activated the two defines:
`// Activate this define to enable a check on the period between zero and the next one

#define CHECK_INT_PERIOD

// Activate this macro to check if all the light were managed in a semi-period

#define CHECK_MANAGED_THYR`

this is my output:, I think the interval between zero-crossings is not correct, although they are fine on scope image. Something with the interrupt I guess:

TLS1000, 2019-09-01:
Also strange: When I set the brightness manually to 255, the library doesn't display the intervals. When I set it lower it seems to do it's thing with the interval detection and so on. Maybe this is something else.

TLS1000, 2019-09-01:
With the verbose parameter set to level 3, this is the serial output:

> Dimmable Light for Arduino: first example
Init the dimmable light class... Done!
setB: posIntoArray:0 pin:25 10000
	raising the light..
No need to shift..
0/0/0
Re-enabling interrupt
E0
	setB: posIntoArray:0 pin:25 197
8638
E0
415
E0
9591
E0
9987
E0
418
E0
9589
E0
9987
E0
416
E0
setB: posIntoArray:0 pin:25 197
No need to perform the exchange, the delay is the same!
0/0/0
	setB: posIntoArray:0 pin:25 197
9592
E0
9986
E0
E0
9986
E0
E0
setB: posIntoArray:0 pin:25 197
No need to perform the exchange, the delay is the same!
0/0/0
	setB: posIntoArray:0 pin:25 197
9987
E0
E0
9987

>

fabiuz7, 2019-09-01:
E0 means that no thyristor is activated, this means that interrupt that performs thyristor activation isn't triggered. 
I think there is a problem of timers/interrupts. Have you check #4? A user meets problems with esp32 rev0: interrupts weren't triggered by timer.

TLS1000, 2019-09-02:
indeed I'm working with ESP32 rev0.... The proposed solution here (adding a yield() ) doesn't solve the problem for me: [https://github.com/espressif/arduino-esp32/issues/1313#issuecomment-423508263]
Calling the timer enable twice does seem to enable the timer, but still the dimmer doesn't work properly:

> timer->dev->config.alarm_en = 1;
	Serial.print("isEnabled: "); Serial.println(timerAlarmEnabled(timer));
	timer->dev->config.alarm_en = 1;
	Serial.print("isEnabled: "); Serial.println(timerAlarmEnabled(timer));	

>

TLS1000, 2019-09-02:
Instead of adding the 'yield()' function before enabling the timer (which causes a crash in my case), adding a 'delayMicroseconds(0)' just before timer enable command works and arms the timer, but still code doesn't execute correct. The startTimerAndTrigger() loop seems to be called multiple times between half cycles:

> 0
isEnabled: 1
9581
isEnabled: 1
isEnabled: 1
isEnabled: 1
417
E0
isEnabled: 1
9581
isEnabled: 1
isEnabled: 1
isEnabled: 1
isEnabled: 1
isEnabled: 1
isEnabled: 1
417
E0
isEnabled: 1
9580
isEnabled: 1
isEnabled: 1
isEnabled: 1
isEnabled: 1
isEnabled: 1
420
E0
isEnabled: 1
9575
isEnabled: 1
419
E0
isEnabled: 1
9579
isEnabled: 1
isEnabled: 1
isEnabled: 1
27
E0
isEnabled: 1
9971
isEnabled: 1
417

> 

>

fabiuz7, 2019-09-02:
Unfortunately I haven't esp32 rev 0 to replicate the bug, but honestly, I would suggest trying another (updated) board (i.e. Lolin D32)

TLS1000, 2019-09-02:
all the doit V1 boards I received via aliexpress(a few days ago) are REV0. And I just ordered 20 ESP's to spread all over my house and control every single light-group... I'm trying now to make something myself based on interrupts. The sync pin triggers the interrupts correctly already, now figuring out how to trigger the timer.

TLS1000, 2019-09-06:
Hi Fabiuz7, with some blood and sweat I managed to get a dimmer working on the ESP32 REV0 board, based on pin and timer interrupt. Key to success is using the timerstart and timerstop functions of the timer class.  Maybe you can add these timer function calls to your library, and make it universal for ESP32 REV0 and REV1? The RobotDyn dimmer library suffers the same issue so your library would be the only one that works. ESP32 sold via Aliexpress (quite a few) are still REV0 versions. I would be happy to test your updated library on a REV0 board.

`#include <Arduino.h>

hw_timer_t * timer = NULL;
volatile uint32_t lastMicros;
volatile int triacDelay=7500;

void IRAM_ATTR isr() {
    
    uint32_t Now=micros();
    if(Now-lastMicros>7000) { //10ms between zero crossings @50Hz, 7ms treshold to avoid false detections does the trick
      digitalWrite(25,LOW);     //stop pulse upon zero crossing
      lastMicros=Now;
      timerAlarmWrite(timer,triacDelay,true);
      timerAlarmEnable(timer);
      timerStart(timer);  
    }
}

void IRAM_ATTR onTimer(){
  digitalWrite(25,HIGH);  //start pulse to ignite lamp after timer
  timerStop(timer); 
}

void setup() {
    Serial.begin(115200);
    
    pinMode(26, INPUT); //zero cross input
    pinMode(25, OUTPUT);  triac output
    attachInterrupt(26, isr, RISING);
    timer=timerBegin(0,80,true);  //although starting and stopping of the timer is done manually, it needs to be autoreload setting
    timerAttachInterrupt(timer,onTimer,true);
    
}

void loop() {
    if (Serial.available())
  {
    int buf = Serial.parseInt();  //accept 0-100 to set the dim level
    if (buf > 0 && buf <=100) triacDelay=10000-100*buf; //calculate delay in µs
    delay(200);
  }
  
    delay(50);
    
}`

this is the dimmercode that works on ESP32 REV0:

fabiuz7, 2019-09-29:
Finally I managed some time to check your solution.

Here your code a bit revised, I can confirm that it works even on ESP32 REV1. 

```
#include <Arduino.h>

hw_timer_t * timer = NULL;
volatile uint32_t lastMicros;
volatile int triacDelay = 7500;

const int ZERO_CROSS_PIN = 23;
const int DIMMER_PIN = 4;

void IRAM_ATTR isr() {
  uint32_t Now=micros();
  if(Now-lastMicros>7000) { //10ms between zero crossings @50Hz, 7ms treshold to avoid false detections does the trick
    digitalWrite(DIMMER_PIN,LOW);     //stop pulse upon zero crossing
    lastMicros=Now;
    timerAlarmWrite(timer,triacDelay,true);
    timerAlarmEnable(timer);
    timerStart(timer);  
  }
}

void IRAM_ATTR onTimer(){
  digitalWrite(DIMMER_PIN,HIGH); //start pulse to ignite lamp after timer
  timerStop(timer);
}

void setup() {
  Serial.begin(115200);
  pinMode(ZERO_CROSS_PIN, INPUT); //zero cross input
  pinMode(DIMMER_PIN, OUTPUT); // triac output
  attachInterrupt(ZERO_CROSS_PIN, isr, RISING);
  timer=timerBegin(0,80,true);  //although starting and stopping of the timer is done manually, it needs to be autoreload setting
  timerAttachInterrupt(timer,onTimer,true);
}

void loop() {
  if (Serial.available())
  {
    int buf = Serial.parseInt(); //accept 0-100 to set the dim level
    if (buf > 0 && buf <=100){
      triacDelay=10000-100*buf;
      Serial.println(triacDelay);
    }
    delay(200);
  }
  
  delay(50);
}
```
I push the upgrade as soon as possible

fabiuz7, 2019-09-29:
I pushed the fix (hopefully). I tried to understand your code, I think that the general idea is that timer must be stopped to be configured (set counter, set alarm,... ).

Might you test it on your board? Again, thanks for the contribution.

TLS1000, 2019-10-23:
hi Fabiuz7, yes I can confirm your code above works on ESP32REV0, I checked the library but in arduino IDE the lib manager says I have the latest version 1.2... no update?

fabiuz7, 2019-10-23:
Actually I'm finishing to review my code and the relative documentation. I plan to release version 1.3 on Saturday.

fabiuz7, 2019-10-27:
New version was published on Arduino :)

closed 2019-10-27 by fabiuz7