fabianoriccardi/dimmable-light #9: use mosfet not triac

ageurtse, 2019-09-02:
Where do i alter the software so i can use a mosfet instate of a triac. 
on ZCD detect i need to set high the light pin, after time elapes the light pin needs to go low. 

not hard in my opinion, but a nice to have futher.

first i go see what the output displays on my scoop.

fabiuz7, 2019-09-02:
Hi,
Your question is not very clear to me, but the core logic is implemented in 2 interrupt functions. You can find the implementation linked below:

- https://github.com/fabiuz7/Dimmable-Light-Arduino/blob/master/src/thyristor.cpp#L79
- https://github.com/fabiuz7/Dimmable-Light-Arduino/blob/master/src/thyristor.cpp#L145

ageurtse, 2019-09-02:
try to make more sense, a triac is fired for a few 4ms after a sertain time to turn on the triac. 
the soner you turn on the triac, the lighter the light lits. 
this is trealing edge dimmer.

i'm trying to build a leading edge dimmer, with 2 mosfets back to back. 
so on ZCD the mosfet nets to turn on, and stay on until the time elapsed then for the rest of the periode it needs to turn off. 
the longer the mosfets are turnd on, the lighter the light lits. 

i hope you understand it.

fabiuz7, 2019-09-02:
To summarize, you would keep the signal HIGH until the end of the semiperiod, right? Because actually the signal to trigger the triac is just a pulse (duration: 15microsecond).

ageurtse, 2019-09-02:
thats right, but when i have time this week, i will try your code and view the output in relation to my ZCD circuit. 
if that is fine i will look into your software to rewrite it and do the above part. 

i only need one channel. 

later i think i would like the ability to invert the signal. so i can switch between trailing and leading edge, then you have a real multifunctional dimmer.

fabiuz7, 2019-09-02:
Yeah this would be a smart improvement! If you need help to understand I will try to help

ageurtse, 2019-09-02:
Thank, i keep you informd

ageurtse, 2019-09-03:
Trying to run your example sketch on my zcd triac dimmer board from dynamic. 

on my own build zcd triac dimmer with a dedicated attiny for dimming the light isn't flickering. 
so it's not the light bulb that is the issue, it's more a software thing. 

what could it be ?

fabiuz7, 2019-09-03:
Those pins are fine. Are zero cross interrupts triggered properly?

ageurtse, 2019-09-03:
I think, how can i check

fabiuz7, 2019-09-04:
You can enable these 2 constants in thyristor.cpp
#define CHECK_INT_PERIOD
#define CHECK_MANAGED_THYR

ageurtse, 2019-09-04:
i try and post the output here.

ageurtse, 2019-09-04:
this is the serial output of the ESP32, please explain what i see

`9425
18
E0
564
E0
9427
18
E0
9982
18
E0
573
E0
9427
17
E0
4
E0
3
E0
572
E0
37
E0
9397
17
E0
556
E0
37
E0
9389
18
E0
570
E0
9427
18
E0
564
E0
9429
17
E0
555
E0
9427
17
E0
564
E0
9430
572
E0
`

fabiuz7, 2019-09-04:
E+{{n}} means how many thyristor are managed for each period. E0 means that your thyristor(s) weren0t activated.
The other numbers are the period passed between a ZeroCrossDetection interrupt and the next one. It seems that there is a problem on your electrical network (or your circuit is subject to noise). I would expect values near to 100 (ms) @50Hz

ageurtse, 2019-09-04:
is there a way to lets say turn of the interupt for 5000 us, to prevent false interupts ?

fabiuz7, 2019-09-04:
yeah, there is the FILTER_INT_PERIOD constant. I test it over multiple boards and found out the 50us is enough. If you would increase this value you are free to try it, however, I think that the circuit may be improved.

I have read that issue: a digital circuit shouldn't be fed with "analog" signals.

ageurtse, 2019-09-04:
do you know a good simple digital ZCD circuit for 230VAC

fabiuz7, 2019-09-04:
It may help, but actually I don't have too much experience to give suggestions.
However, In repository description I've linked a board equipped with 8 thyristors (the seller has many available thyristor circuits). I've observed the ZCD output with an oscilloscope and it has squared edges

ageurtse, 2019-09-05:
The hex schmitt trigger worked. 
but now i think i don't going to use your code, because my code is also working. the problems i had was false interupts. 

attached the code what i use. 
It's all hardware driven, so no worry about timmings and that kind of stuff, i love the esp32
`#include "driver/mcpwm.h"

#define GPIO_SYNC0_IN 22       //pin for the external interrupt (Input with Pullup)                         
#define MOSFET 23          //pin to trigger the MOSFET (Output) pin22

void setup() {
  Serial.begin(115200);
  // put your setup code here, to run once:
  mcpwm_pin_config_t pin_config; 
    pin_config.mcpwm_sync0_in_num  = GPIO_SYNC0_IN;
      
  mcpwm_config_t pwm_config;
    pwm_config.frequency = 100;    //frequency = 110Hz
    pwm_config.cmpr_a = 100;       //duty cycle of PWMxA = 60.0%
    pwm_config.counter_mode = MCPWM_UP_COUNTER;
    pwm_config.duty_mode = MCPWM_DUTY_MODE_0;
    
  mcpwm_set_pin(MCPWM_UNIT_0, &pin_config);
  mcpwm_gpio_init(MCPWM_UNIT_0, MCPWM0A, MOSFET);
  mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_0, &pwm_config);   //Configure PWM0A & PWM0B with above settings
  mcpwm_set_duty_type(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_A, MCPWM_DUTY_MODE_0);
  mcpwm_start(MCPWM_UNIT_0, MCPWM_TIMER_0);
  mcpwm_sync_enable(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_SELECT_SYNC0, 990);
}

void loop() {
  mcpwm_set_duty(MCPWM_UNIT_0, MCPWM_TIMER_0, MCPWM_OPR_A,35);
  delay(1000);
  Serial.println("fire");
}`

closed 2019-09-06 by fabiuz7