fabianoriccardi/dimmable-light #10: comments

jandrassy, 2019-10-02:
There is a  [TriacDimmer library for AVR](https://github.com/AJMansfield/TriacDimmer), which uses timer capture instead of the external interrupt. I use it in AVR version of my project, which controls a heater.

[Here ](https://arduino.stackexchange.com/questions/63631/fan-regulator-control-from-arduino/63674#63674)I have my AVR Triac test sketches.

For SAMD Arduino I have in my project[ a dimmer library for SAMD architecture](https://github.com/jandrassy/Regulator/blob/master/TriacLib/TriacLib.h).

The use of delayMicroseconds in interrupt is not good. You should set a new timer to end the pulse or use a timer wave output which creates a pulse.

I will try your library with esp8266 for esp8266 version of my heater regulator.

fabiuz7, 2019-10-02:
Hi jandrassy,

> There is a TriacDimmer library for AVR, which uses timer capture instead of the external interrupt. I use it in AVR version of my project, which controls a heater.

I know that there are better ways to implement dimmer control, but some of these depend on microcontroller peculiarity (such as timer capture for AVR).

> The use of delayMicroseconds in the interrupt is not good. You should set a new timer to end the pulse or use a timer wave output which creates a pulse.

You are right, but I decided on that solution for simplicity, however, I was planning to implement this idea. Now I think I will prioritize this optimization, enough general to be implemented for each microcontroller.

I will look to your repo for further hints, thanks for the feedback.

   Fabiano

fabiuz7, 2019-10-16:
I had implemented your suggestion about the interrupt for esp8266 and esp32. Hopefully I will finish the AVR tomorrow, then I will push the modification.

fabiuz7, 2019-10-17:
I had pushed the update for avr, esp8266 and esp32. My old solution is still available, but by default is disabled (#define PREDEFINED_PULSE_LENGTH is commented).

If you test this last commit, please let me know

jandrassy, 2019-10-19:
I need at least 10 bit resolution.  I regulate from 300 to 2050 W. 8 W per step is not fine enough. The TriacDimmer AVR library and my SAMD library use infinite 0 to 1 float interval.

fabiuz7, 2019-10-19:
I would avoid float type for performance reasons. Is it enough microsecond resolution? 

Thyristor class use microsecond and it is at the basis of this library. DimmableLight class is just a wrapper over Thyristor class.

jandrassy, 2019-10-19:
yes, microsecond are good

I tested your library with Robotdyn AC dimmer with Arduino Nano and with Wemos D1 mini with a 40 W light bulb. it works.

Tomorow I will test the regulation of the heater with your library on ATmega 1284p board currently in project.

fabiuz7, 2019-10-27:
New version 1.3.0 that includes interrupt optimization is published on Arduino :)

closed 2019-10-27 by fabiuz7

jandrassy, 2019-10-27:
I decided to use[ my own code](https://github.com/jandrassy/Regulator/blob/master/TriacLib/TriacLib.h) on AVR. It has only zero crossing interrupt which resets the Timer1 counter to 0. The Triac pin is set HIGH by Timer1 on OCR and toggled off on Timer top. The Timer runs Fast PWM at almost 100 Hz. To sync the Timer counter is set to 0 in zero crossing interrupt.

for esp8266 I will see once I have time for it