fabianoriccardi/dimmable-light #12: Flickering and odd behaviour

RobertSmart, 2019-11-05:
Hi,

Trying to use this on an ESP32. Using is on a 60 watt incandesent bulb, using the basic dimmable light example code. I'm using the standard robodyne triac dimmer module.

I can't seem to get it to work correctly as its very flickery and switches on and off.

I've tried setting the frequency to 50hz (I'm in the UK) by putting this at the top of the page #define NETWORK_FREQ_50HZ, but its still the same. 

Am I missing something?

Thanks,

Rob Smart

fabiuz7, 2019-11-05:
Hi Robert,
thanks for the feedback, I will investigate as soon as possible! Could you tell me which version of ESP32?

Have you tried older version the library such as v1.3.0?

fabiuz7, 2019-11-05:
I had tested again v1.4.0 downloaded from Arduino IDE, uploaded the first example and it works perfectly (syncPin=23; thyristorPin=4). I'm using Lolin D32 board and the board linked in the readme.

I've also tested with yours board (i think it is https://www.amazon.com/GENUINE-RobotDYN-Programmable-Controller-Compatible/dp/B071X19VL1/ref=pd_sbs_147_t_0/142-4102923-7105853?_refRID=1J65VTG5SK36047QWEKN), and it flickers because the zero_cross_interrupt is dirty. 

You can uncomment FILTER_INT_PERIOD and PREDEFINED_PULSE_LENGTH at the begin of file thyristor.cpp

RobertSmart, 2019-11-06:
Hi,

That is the same board I'm using. What do you mean by the zero cross interrupt is dirty?

I’ll try editing the file and trying again as you suggested.

Thanks

Robert Smart MSc

fabiuz7, 2019-11-06:
I've taken a photo to show you the difference between dimmer boards: each one has different Zero Cross Detection Signal.
The higher line comes from Robotdyn, the second comes from Krida Electronics (you can notice it is sharper and shorter pulse).

[image]
Legend: (X: 100us/div; Y: 1V/div).

RobertSmart, 2019-11-07:
OK, I'll try it now. Do I need to change anything to make is use 50hz?

Thanks,

Rob Smart

fabiuz7, 2019-11-07:
By default it is set to work at 50Hz

fabiuz7, 2019-11-12:
Hi Robert, 
did it work?

RobertSmart, 2019-11-12:
Hi, Yes it did thank you.

closed 2019-11-16 by fabiuz7

nickrugge, 2020-04-30:
Ciao, 
innanzitutto complimenti per il fantastico lavoro fatto su questa libreria, è l'unica tra le varie provate, inclusa quella del produttore, che lavora come si deve con la dimming board della robotdyn.

Sto utilizzando una dev board esp8266 per controllare attraverso mqtt due lampade in dimming. (Ho la board robotdyn con due output) ZeroCross: GPIO4, DMR1: GPIO 5, DMR2: GPIO16.
Nel firmware utilizzo la libreria ArduinoOTA e per la comunicazione mqtt uso la libreria AsyncMqttClient. Inoltre durante il loop eseguo il digitalRead di 4 ingressi con un minimo di logica.
Attualmente il server mqtt è offline e quinidi la board si limita a eseguire il tentativo di connessione. 

Utilizzando la board robotdyn ho dovuto da subito definire la variabile FILTER_INT_PERIOD come hai conisgliato per via del costante flickerin; define che ha migliorato di gran lunga la situazione.
Mi ritrovo comunque ogni tanto con le luci che flickerano per lassi di tempo e intensità variabile anche con luminosità prossima al 100%. Impostandola al 100% invece il flickering ovviamente smette.
Se definisco la variabile PREDEFINED_PULSE_LENGTH la situazione peggiora.
A cosa potrebbe essere dovuto questo comportamento? 
Forse la gestione di 2 tiristori, zero cross detection e il timer interrupt per il tentativo di connessione sono troppo per un esp8266? Che prove mi consiglieresti di fare? Purtroppo non dispongo di un oscilloscopio quindi non posso verificare cosa accada a livello hardware. Ho provato a sostituire diversi tipi di lampade (sia a led che a incandescenza) ma il risultato è esattamente lo stesso.
Nel frattempo ho ordinato una esp32 su cui proverò la stessa configurazione e ti farò sapere.

Grazie
Nick

fabiuz7, 2020-05-01:
Ciao Nick, 
hai provato che i dimmer e la libreria funzionino correttamente? Per esserne sicuro proverei quindi a rimuovere i tentativi di connessione alla rete per verificare se il problema sta nella mia libreria o nell'integrazione con la parte "wifi". Il problema sicuramente non sta nella lettura digitale dei 4 pin.
Flickerano entrambe le lampade allo stesso modo?

nickrugge, 2020-05-03:
Ciao e grazie, prima di eseguire la prova che mi hai consigliato ho provato per curiosità ad aumentare i seguenti valori:

```
semiPeriodShrinkMargin = 100;
semiPeriodExpandMargin = 100;
```

Sono due giorni che tengo le lampade controllate e non ho ancora notato flicker.
Oltretutto adesso la connessione con il server mqtt è attiva e anche durante lo scambio di messaggi non ci sono problemi, pare quindi che la gestione di wifi e mqtt non influisca per nulla.
Probabilmente il segnale di zero cross era ancora troppo sporco con 50us di margine.
Toglimi una curiosità: agire sui valori startMargin e endMargin sarebbe stato lo stesso?

fabiuz7, 2020-05-03:
Perfetto, son contento che sia riuscito a mettere a posto. 

In realtà i le 2 coppie di parametri rispondono a 2 necessità diverse:  il filtraggio (quello che te hai applicato) serve ad evitare che scattino interrupt nei momenti sicuramente sbagliati (interrupt troppo vicini tra loro). 

Invece startMargin e endMargin servono ad evitare che il segnale di attivazione che invii da arduino arrivi troppo vicino allo zero cross, momento in cui il dimmer fa più fatica ad attivarsi. 

Diciamo che forse il tuo problema **potrebbe** essere risolto tramite incremento di startMargin e endMargin, ma dipende dal tipo di rumore che hai sulla linea elettrica di casa.