fabianoriccardi/dimmable-light #15: esp32 problem

kingkimohero69, 2020-01-04:
I use ESP32 with a dimmer built by me similar to RobotDyn dimmer uses MOC3021 and 4N25,
everything works fine except two things:
1-when i put brightness.set(0), my bulb shows a faded light ,also my fan has a strange small noise.
2- The dimmer does not start until i pull up and put in the electricity socket a couple of times.

closed 2020-01-23 by kingkimohero69

fabiuz7, 2020-01-27:
I @kingkimohero69,
may I ask you if you had solved the problem?

kingkimohero69, 2020-02-07:
I bought another circuit it works now perfect, you did mention to use the filter with robodyn it works great except when i am close to 92% it flickers , i solved buy not passing it :)
Another problem with my Ceiling fan it makes humming noise ,After a lot of search i figure out to use capacitors and some relays instead of dimming by voltage

jammer99, 2021-07-01:
@kingkimohero69 i am also having humming noise problem. could you point me in the right direction on how to use capacitors and relays to remove the noise?

kingkimohero69, 2021-07-03:
@jammer99 the circuit is the same as the ordinary speed control switch which comes with the ceiling fan, except you will replace the mechanism part with relays so if the switch comes with 1,2,3,4,5 speed level you will make a circuit with 5 relays and ESP and you can add the six one for turning off and attach it to the switch box.

The URL shows the circuit for the ordinary control circuit to a ceiling fan with capacitors, you will have to attach relays between switch knob wire and med wire, high wire, low wire, of course, you can control them with the ESP.[https://i.pinimg.com/736x/62/e6/63/62e663b2849f80e2de30bb47169b07b7.jpg](url)