fabianoriccardi/dimmable-light #16: ESP32 w/OTA - Guru Meditation Error: Core 1 panic'ed

dominatorstang, 2020-01-17:
First let me say, THANK YOU for this library build!  No other library gives me such a clean output to my LED lights with my ESP32 and RobotDyn Dimmer

Okay, here is what I am experiencing.  When uploading new code using the OTA web interface, it gets to 13% or so and freezes.  I then had the serial port open in Arduino when I tried the web interface and noticed I get "Guru Meditation Error: Core 1 panic'ed (Cache disabled but cached memory region accessed)".

This ESP32 is handling a fair amount of code and devices so I used a spare ESP32 and did some bench testing.  I have it down to only the AC Dimmer, this library and the OTA library while still having the issue.

I did also figure out that if I have the light turned off, then the OTA update works fine.  So for now I have been making sure I turn the light off before I send an update to it.  I had also noticed after sending code over the USB connection, it would sometimes crash and restart a time or two on the first boot.

I plan to try more troubleshooting over the weekend to give more details.

dominatorstang, 2020-01-18:
Perhaps the OTA library locks the cache when uploading the program, but then the zero cross interrupt is firing with the light on and using the locked cache.  The link below looks similar to what I am experiencing.  I am still researching to understand how to resolve it.
[https://github.com/espressif/arduino-esp32/issues/855](url)

Arafa-microsys, 2021-04-08:
@dominatorstang Are you solve this problem?
I am facing the same problem but hw_interrupt instead of OTA, with SPIFFS.
When I make any Operation based on SPIFFS the same crash happens.
Please, if you overcome the problem tell me the solution.

dominatorstang, 2021-04-08:
Sorry but I just make sure the light is off before I update.  I had not looked Any further.

Arafa-microsys, 2021-04-10:
> Sorry but I just make sure the light is off before I update. I had not looked Any further.

Yes, Actually when turning on the light at full brightness or turn off the light completely, the hardware timer stop by the ActivateTriac function because called StopTimer(), therefore the problem temporarily solved.
Thank you for your reply.
We are looking for a solution without turning off the lights.