fabianoriccardi/dimmable-light #17: Add feature to measure frequency

usrrsr, 2020-04-28:
Hello , Thanks for Great library.
I request you to add function to measure the duration between two consecutive zero crossing interrupt and determine the frequency based on this interval.

regards

fabiuz7, 2020-04-28:
Hi, thanks!
At the time I was thinking about it, but finally, I didn't spot a meaningful purpose if not debugging and auto-detected the electrical network frequency. 

Have you another use case for this feature?

usrrsr, 2020-04-29:
Yes, i want to use it for motor controller so i want to know frequency well before controlling the speed. Also in such case it is desirable to add function to library like getfrequency() to get it. During calling this function one can turn off dimming functionality to be on safe side

RudyFiero, 2020-04-29:
I have suggested that our programmers default to 60Hz until it can be determined that 50Hz is being used. This way the output will be safe for the motors, and lights will not flicker. This allows non-delayed operation.

fabiuz7, 2020-05-01:
As already said, I thought to implement an "autodetect frequency" feature: the library can wait for 100ms in order to detect the network frequency. But what if there are spurious noise on the network? The measurement can be affected, and the library may detect the wrong frequency.

About @usrrsr point, what should *getfrequency()* return? A number or it should discriminate only between 50hz and 60Hz?

usrrsr, 2020-05-01:
Preferably it should return average number of 3  to 4 or user specified samples

fabiuz7, 2020-05-10:
Hi, sorry for the late reply, they seem good suggestions. This week I will try to find out some moments to finalize them!

fabiuz7, 2020-05-17:
I've started to push the first enhancements, by this evening I also will add an example to show the usage and a bit of documentation. The code is not in its final version, I have to debug and test for all the platform (actually I'm developing on ESP8266). Also, I cannot test the 60Hz version, hence I would ask you a feedback on that.

usrrsr, 2020-05-18:
> I've started to push the first enhancements, by this evening I also will add an example to show the usage and a bit of documentation. The code is not in its final version, I have to debug and test [...]

Thanks a lot
Its a highly required improvement

fabianoriccardi, 2023-11-09:
Closing since, finally, the monitor frequency feature was released in [v1.5.0](https://github.com/fabianoriccardi/dimmable-light/releases/tag/1.5.0).

closed 2023-11-09 by fabianoriccardi