fabianoriccardi/dimmable-light #21: ESP32 Flicker solution

jlhavens, 2021-01-07:
Thanks for the work on this project.  I can see that you expended a lot of time and effort.

When I tried your library, I noticed some flicker in a regular incandescent bulb.  (I also got a compiling error because of a missing ';'  in  dimmable_light_linerizable.h  line 62 .

While investigating a slightly different approach for an ESP32 project, I discovered that my zero crossing interrupt fired on both the rising and falling edges, even though it was set to fire on the RISING.  I discovered this by measuring the interval between interrupts.  Some were 8300 microseconds and some were 475 microseconds.  I didn't seem to get any that were 7800.

I made the following changes to thyrister.cpp, which pretty much eliminated the flicker.  This was only on a simple demo platform so I don't know if any other causes my exist.

   #elif defined(ARDUINO_ARCH_ESP32)
   void IRAM_ATTR zero_cross_int(){
   // inserted 
   static unsigned long now = 0;
   if(micros()-now < 1000) return;
   now = micros();
   //end inserted
   #else

dominatorstang, 2021-04-17:
I notice this same issue and research shows it is a common issue for ESP32.  I have tried multiple ESP32 boards and all give the same result of trigger on rising and falling edge.  Others have been able to prove that the speed in which the signal rises and falls will decide if this issue happens of not.  Adding a capacitor to try and filter the signal will almost certainly cause even more false triggers because of the slower signal transition.

I tried tweaking the resistor size for the pullup and did help reduce the false triggers, but did not eliminate.

dominatorstang, 2021-04-19:
My zero cross signal from a RobotDyn AC light dimmer is taking 250us to rise and about the same to fall.  The 4N25 IC that RobotDyn used specs to be much quicker than this.  Not sure if it is just the design of the circuit or a cheap knock off that is causing this.

At the moment I have an Arduino UNO taking the zero cross signal in from the RobotDyn and spitting back out a nice sharp signal to the ESP32.  I also have a Schmitt trigger on its way that will be used to do the same thing for the long term.

P.S. I also noticed some false triggers on the ESP32 interrupt when using pin 25 or 26 for interrupt and reading analog on 36,39,34,35,32 or 33.  Another thing to note that I have not completed troubleshooting on, but it appears that a select few pins being used for PWM will cause false interrupt triggers.

pabloeorsi, 2021-09-11:
Hi! Thank you first for sharing!
I am having a flickering problem on my ESP32 DEVKIT V1. It was randomly eliminated and now it started again. I already did a lot of hardware tests and I think there are no problems. It seems to be a problem with interrupts but I can't figure it out.
Could they solve the problem? I would appreciate your help. Greetings!

dominatorstang, 2021-09-11:
They may be a solution for you, yes.  Do you have an arudino uno, micro, or mini?  If so, you can always create a little program to take in the slow logic curve from the dimmer and put out a nice sharp logic curve to the ESP32 for testing.

I also recommend enabling the below in the thyristor.cpp file so you can see the false triggers in the serial output of the ESP32.
//#define PRINT_INT_PERIOD

ssarna1, 2023-03-12:
I had the same problem, i have connected the scope and set up an output which toggled every time a crossing was detected - it turned out that sometimes the zero crossing was actually detected on both rising and falling edges, and sometimes even twice during one of the edge. i have modified line 37 of thyrystor.cpp as mentioned above, uncommented:
#define FILTER_INT_PERIOD

but that was not enough, i also had to increase variables in lines 224 and 225 (defaults were 50):
const int semiPeriodShrinkMargin = 300; 
const int semiPeriodExpandMargin = 300;

i hope it helps someone.

fabianoriccardi, 2023-11-03:
Hi guys, I'm closing this issue since, years later, I found a bit of time to finally investigate the issue by myself. The main (recent) discussion is in #41, but basically all what you said is correct.

In particular what @ssarna1 has suggested is correct, but after some tests of oscilloscope with RobotDyn dimmer I would propose the following values:

``` C
const int semiPeriodShrinkMargin = 400;
const int semiPeriodExpandMargin = 50;
```

If you cannot change or modify the hardware, this is the only acceptable software solution provided by the library.

closed 2023-11-03 by fabianoriccardi