fabianoriccardi/dimmable-light #23: Crashes on the ESP32 with SPIFFS

Arafa-microsys, 2021-04-08:
I am facing a crash when using SPIFFS with Library.
The SPIFFS locks the cash when used and the timer interrupts still working, So the problem happened.
I search for this problem and found a solution in ESP32 Forum It is summarized in the following:
adding DRAM_ATTR to ISR maybe fix this problem, But I can't apply it because DRAM_ATTR  should apply on Static Variables Used by ISR As far as I understand.

Arafa-microsys, 2021-04-08:
> Also inside the ISR context, unless data is tagged as DRAM_ATTR it is possible to crash the esp32. If it is tagged it is read-only. The only data that can be accessed as read-write is inside IRAM and only accessible as 32bit access (byte access will cause a panic).
https://github.com/bakerstu/openmrn/issues/290#issuecomment-483093795

Arafa-microsys, 2021-04-08:
> IRAM is very limited and we need to be very selective as to what we tag as that. We shouldn't have any ISRs in OpenMRNLite code for the esp32, except in the load test example which uses a hardware timer which uses an ISR context for the callback invocation. Outside of this usage there shouldn't be any other ISR usages at this time.
> 
> One side note on the esp32, using the IRAM_ATTR tag does _NOT_ make it safe to use SPIFFS! If you access spi_flash it will crash almost everytime with an error related to "cache accessed while cache disabled". For this reason, I would avoid ISRs/timers on the esp32 unless absolutely necessary.

https://github.com/bakerstu/openmrn/issues/290#issuecomment-483093335

simonchen007, 2021-05-31:
I face to the same issue, have you got resolution yet ？

simonchen007, 2021-05-31:
it seems the same even if I remove all IRAM_ATTR tag. So, it seems this problem is not because the IRAM tag.

simonchen007, 2021-05-31:
I found the issue, thing is , you can run "light.setBrightness(i)" after everything initialization done.
eg:
```c
 if (millis() > 8000) {
light.setBrightness(me.heater_power);
delay(50);
}
```

fabianoriccardi, 2023-11-09:
I don't think this is related to IRAM_ATTR, since the interrupts are registered without asking to ESP-IDF to disable flash interrupts. However, if it still matters, can you post an MCVE and the version you were working with?