fabianoriccardi/dimmable-light #24: More than 8ch

kleurbleur, 2021-08-14:
I'm using three RobotDyn 4ch dimmer boards all getting their power from a MeanWell 5V 3.4A power supply. Since they're connected to the same 230V 50Hz power outlet they also all have their zero-cross connected to the Ardiuno Uno GPIO 2. 

In total I have 12 channels and if I mix the channels around spread across the boards the first 8 of the list work. I tested this by putting the last not working 4 channels in the beginning of the list and vice versa. Result was that the ones that did not work worked, and the others didn't. 

Now I checked the source code and changed the on line 95 in file dimmable_light.h from 8 to 12, but to no avail. Does anybody know how to raise the limit of the maximum amount of channels?

kleurbleur, 2021-08-15:
I have to run some extra tests, but the test above seems to run good now. I raised the number of maximum lights to 12 by adjusting  `static const uint8_t N = 8;` to 12 on line 91 in thrystor.h

fabianoriccardi, 2023-03-21:
Technically there is no limit (leaving out the memory and pins) to the number of dimmers. 

Just change the "soft" limit to the desired value here:
https://github.com/fabianoriccardi/dimmable-light/blob/c7803314f92708f895227ede46426f68ac7863e4/src/thyristor.h#L165

and here:

https://github.com/fabianoriccardi/dimmable-light/blob/c7803314f92708f895227ede46426f68ac7863e4/src/dimmable_light.h#L129

bcelary, 2023-03-21:
Glad to hear that. Maybe it is worth adding this in a comment so when others look at the limitation are going to be less concerned with making a change. Or perhaps maybe make it adjustable at the compile time?

I have cloned the repo anyway as I think I might need to also be able to retrieve the pin number from the thyristor class. This is handy when I have an array of dimmers.

Mangata42, 2023-11-01:
Hey @kleurbleur @bcelary , did you manage to go up to 16 lights?
I've changed the lines mentionned in the lib but to no success, I'm having a lot of flickering issues above 8 lights. It's worth mentionning that I'm using a 5v5A power unit. 
Thanks!

kleurbleur, 2023-11-01:
I did not test it above 12, but flickering can happen if the zero cross is not good. 
Are all the lamps the same? 
Do they use the same power source? 
What happens if you swap them around?

Mangata42, 2023-11-01:
@kleurbleur Wow, first of all thank you for your fast answer even after 2years! 
The lamps are all the same indeed, and they use the same power source. Swapping them around does not change anything. 
I think that it might be the fact that only one RoboDyn dimmer board is connected to the Zero-Cross pin of the Arduino, but overall it's a mechanism with 4 dimmer boards all connected to one Arduino. I should try this tomorrow.

kleurbleur, 2023-11-01:
How I read my opening post I all different boards should have their zero-cross connected to the same Arduino pin indeed. Otherwise the board doesn't know when to do the dimming and you get flickering.

fabianoriccardi, 2023-11-01:
Which board are you using? 
I don't think using a single ZC pin is an issue, since I guess your lamps are all connected to the same AC network, hence they share the same AC frequency and phase.

Do the lamps work properly with a single board (4 lamps)?

Mangata42, 2023-11-01:
I'm using the RoboDyn 4CH board. My lights are all on the same AC indeed. 
Yes, your code works perfectly with up to two dimmable boards together, so 8 lights, but above that, it flickers an awful lot. 
I'm trying to achieve 4 dimmable boards, so 16 lights bulbs all controlled by one Arduino. 
I might try connecting all of the zero cross to the Arduino but logically speaking I guess one is enough. 

Thank you for your help :)

fabianoriccardi, 2023-11-01:
That's fine. Which Arduino board are you using? Arduino UNO, Arduino mega, or other? I need to know at least the generic platform you are working with. 

Moreover, if you control only 9 lamps, do they run smoothly?

Mangata42, 2023-11-01:
Sorry forgot to mention the board!
I'm using Arduino Mega 2560. It works fine with 8 but starts flickering with 9. Might be software related?

fabianoriccardi, 2023-11-01:
I guess, but I have to give it a closer look.

Mangata42, 2023-11-01:
I'll try connecting all of the Z-C to the Arduino tomorrow and see if that changes anything. I'll keep you updated.
Thank you, you're a saint.

fabianoriccardi, 2023-11-01:
For now, I can tell I can replicate issue.

Mangata42, 2023-11-01:
Awesome. Let me know if you need anything.

-------- Message d'origine --------

Mangata42, 2023-11-02:
Hello again. 
I just tried to connect all of the zero-cross of each dimmable light boards to the Arduino, but still the same issue unfortunately above 8 lights, violent flickering. 
@kleurbleur How did you manage to go up to even 12? I feel like the only difference in our circuitry would be the Arduino (I have a mega) but still. 
Thanks!

kleurbleur, 2023-11-02:
I was using an ESP32 so maybe that's the difference? But this is al a long time ago for me and the company I made it for got bankrupt in the end so I salvaged most parts and took my loss. I don't have documentation of it I'm afraid.

Mangata42, 2023-11-02:
@kleurbleur Holy shit lol that's a story on it's own. It's alright, thank you anyway! Hope you're doing good now.

fabianoriccardi, 2023-11-04:
May you post a MCVE sketch that makes the lamps flicker?

Mangata42, 2023-11-04:
> May you post a MCVE sketch that makes the lamps flicker?

Sure, here you go: [Gist Code](https://gist.github.com/Mangata42/d4c61943a2ac82347ac37cd37dd7809b )

fabianoriccardi, 2023-11-04:
This is not exactly an MCVE since it requires the device that sends the messages through Ethernet.

However, I have investigated a bit and I have found an error that introduces the flickering when many dimmers are controlled.

Mangata42, 2023-11-05:
That's great news! Is it something you can fix?

fabianoriccardi, 2023-11-05:
I hope so. I have pushed a possible fix to branch https://github.com/fabianoriccardi/dimmable-light/tree/fix-16-dimmers . Can you test it on your setup?

Mangata42, 2023-11-05:
Thank you Fabiano. Sure, I'll give it a go as soon as I can and let you know how it went. Thanks a lot!

fabianoriccardi, 2023-11-12:
Closing as [v1.5.2](https://github.com/fabianoriccardi/dimmable-light/releases/tag/1.5.2) (which includes the fix) is out.

closed 2023-11-12 by fabianoriccardi