fabianoriccardi/dimmable-light #26: Cannot compile the code for UNO or ESP8266

vks007, 2021-10-14:
Hi, I am not able to compile the example code for any board. Is there an update required for this library or am I missing something. please help.
For Arduino Uno I get the following compile error :
`C:\My Data\OneDrive\Documents\Circuits\Sketches\libraries\ArduinoSTL\src\del_opnt.cpp:25:56: error: 'nothrow_t' in namespace 'std' does not name a type
 _UCXXEXPORT void operator delete(void* ptr, const std::nothrow_t& ) throw() {
                                                        ^~~~~~~~~
"C:\\Users\\U0031929\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\My Data\\OneDrive\\Documents\\Circuits\\Sketches\\libraries\\Dimmable_Light_for_Arduino\\src" "-IC:\\My Data\\OneDrive\\Documents\\Circuits\\Sketches\\libraries\\ArduinoSTL\\src" "C:\\My Data\\OneDrive\\Documents\\Circuits\\Sketches\\libraries\\ArduinoSTL\\src\\del_ops.cpp" -o "C:\\Users\\U0031929\\AppData\\Local\\Temp\\arduino_build_797941\\libraries\\ArduinoSTL\\del_ops.cpp.o"
C:\My Data\OneDrive\Documents\Circuits\Sketches\libraries\ArduinoSTL\src\del_ops.cpp:25:50: error: 'std::size_t' has not been declared
 _UCXXEXPORT void operator delete(void* ptr, std::size_t) throw(){
                                                  ^~~~~~
Using library Dimmable_Light_for_Arduino at version 1.4.0 in folder: C:\My Data\OneDrive\Documents\Circuits\Sketches\libraries\Dimmable_Light_for_Arduino 
Using library ArduinoSTL at version 1.1.0 in folder: C:\My Data\OneDrive\Documents\Circuits\Sketches\libraries\ArduinoSTL 
exit status 1
Error compiling for board Arduino Uno.
`

For ESP8266 I get the following errors:
`C:\My Data\OneDrive\Documents\Circuits\Sketches\libraries\Dimmable_Light_for_Arduino\src\hw_timer_esp8266.c: In function 'hw_timer_arm':
C:\My Data\OneDrive\Documents\Circuits\Sketches\libraries\Dimmable_Light_for_Arduino\src\hw_timer_esp8266.c:34:13: warning: 'RTC_REG_WRITE' macro is deprecated
   34 |     RTC_REG_WRITE(FRC1_LOAD_ADDRESS, US_TO_RTC_TIMER_TICKS(val));
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\My Data\OneDrive\Documents\Circuits\Sketches\libraries\Dimmable_Light_for_Arduino\src\hw_timer_esp8266.c: In function 'hw_timer_init':
C:\My Data\OneDrive\Documents\Circuits\Sketches\libraries\Dimmable_Light_for_Arduino\src\hw_timer_esp8266.c:73:13: warning: 'RTC_REG_WRITE' macro is deprecated
   73 |                       FRC1_AUTO_LOAD | DIVDED_BY_16 | FRC1_ENABLE_TIMER | TM_EDGE_INT);
      |             ^         ~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\My Data\OneDrive\Documents\Circuits\Sketches\libraries\Dimmable_Light_for_Arduino\src\hw_timer_esp8266.c:76:13: warning: 'RTC_REG_WRITE' macro is deprecated
   76 |                       DIVDED_BY_16 | FRC1_ENABLE_TIMER | TM_EDGE_INT);
      |             ^         ~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\My Data\OneDrive\Documents\Circuits\Sketches\libraries\Dimmable_Light_for_Arduino\src\hw_timer_esp8266.c:80:9: warning: 'hw_timer_isr_cb' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
   80 |         ETS_FRC_TIMER1_NMI_INTR_ATTACH(hw_timer_isr_cb);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\My Data\OneDrive\Documents\Circuits\Sketches\libraries\Dimmable_Light_for_Arduino\src\hw_timer_esp8266.c:50:30: note: declared here
   50 | static ICACHE_RAM_ATTR void  hw_timer_isr_cb(void)
      |                              ^~~~~~~~~~~~~~~
C:\My Data\OneDrive\Documents\Circuits\Sketches\libraries\Dimmable_Light_for_Arduino\src\hw_timer_esp8266.c:82:9: warning: 'hw_timer_isr_cb' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
   82 |         ETS_FRC_TIMER1_INTR_ATTACH(hw_timer_isr_cb, NULL);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~
C:\My Data\OneDrive\Documents\Circuits\Sketches\libraries\Dimmable_Light_for_Arduino\src\hw_timer_esp8266.c:50:30: note: declared here
   50 | static ICACHE_RAM_ATTR void  hw_timer_isr_cb(void)
      |                              ^~~~~~~~~~~~~~~
`

vks007, 2021-10-14:
I was able to resolve the issue with Arduino UNO by downgrading the version of IDE to 1.1.12 as suggested [here](https://forum.arduino.cc/t/arduinostl-problem/666354) but the issue for ESP remains.

fabianoriccardi, 2022-08-21:
Sorry for the long delay, I'm trying to catch up with this library. I can confirm the error related to Arduino Uno. Unfortunately, that issue is out of the scope of this library, and it can "workarounded" by using the AVR core v1.8.2 and the latest version of ArduinoSTL. 

About the ESP8266, I can't see any error in your log, just some deprecation warnings that shouldn't create any problem (however I will try to solve them).

fabianoriccardi, 2023-11-02:
Closing since:
- on Arduino: the issue is fixable by downgrading the core to v1.8.2 or deleting the `dimmable_light_manager.*` if you don't need that class.
- on ESP8266: I cannot replicate the issue.

closed 2023-11-02 by fabianoriccardi