fabianoriccardi/dimmable-light #28: Adafruit ESP32 Feather crash on examples

benkuper, 2022-02-21:
Hello, I'm trying this lib after having tried RBDDimmer on ESP32 and seeing that this one seems more promising regarding flickering issue.

I actually can't get it to work the basic examples, using an Adafruit ESP32 Feather, using output pin 12 and zerocross pin 39.

here is the output of the serial monitor :
```
Dimmable Light for Arduino: first example
Init the dimmable light class... Done!
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008116e  PS      : 0x00060e31  A0      : 0x800813fa  A1      : 0x3ffbee1c  
A2      : 0x0000251c  A3      : 0x000000c7  A4      : 0x0000251b  A5      : 0x0000251c  
A6      : 0x000000c8  A7      : 0x00000000  A8      : 0x00010000  A9      : 0x00000000  
A10     : 0x00000001  A11     : 0x400811d0  A12     : 0x3ffb91f0  A13     : 0x3ffb2650  
A14     : 0x3ffc18ac  A15     : 0x3ffb91cc  SAR     : 0x00000020  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0001001c  LBEG    : 0x40085f08  LEND    : 0x40085f13  LCOUNT  : 0xffffffff  

Backtrace:0x4008116b:0x3ffbee1c |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
```

Is there a list of authorized / unauthorized pins for output / zerocross ?

Thank you

benkuper, 2022-02-22:
After further tests, this crash only happens when I'm actually plugging the Krida dimmer board. When nothing is plugged, the program is running, which makes me think it has to do with code that happens when receiving the interrupt

fabianoriccardi, 2022-02-22:
Hi Ben, can you tell me which core version you are using? I didn't tried the most recent ones... Pins 39/12 are fine.

benkuper, 2022-03-02:
I'm on Arduino 1.8.19, using ESP32 version 2.0.2  on the latest commit of this repo.
I think I also tried with PlatformIO, but I guess the ESP version is the same

fabianoriccardi, 2022-03-07:
Thanks for the information. 

Unfortunately I left behind the new versions of Arduino Cores, I have to catch up. I will give a look in the next few days!

D1ver, 2022-08-15:
> Thanks for the information.
> 
> Unfortunately I left behind the new versions of Arduino Cores, I have to catch up. I will give a look in the next few days!

Hi! Do you have an updates that solves this problem?

daliangowert, 2022-11-30:
Hi! I tried to compile the example code for my ESP32 which is connected to the RobotDyn Dimmer DIN 4CH module and also got the same reported:

> Dimmable Light for Arduino: first example
> Initializing the dimmable light class... Done!
> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
> 
> Core  1 register dump:
> PC      : 0x4008140e  PS      : 0x00060531  A0      : 0x800813de  A1      : 0x3ffbeddc  
> A2      : 0x00001e99  A3      : 0x000000c7  A4      : 0x00001e98  A5      : 0x00001e99
> A6      : 0x000000c8  A7      : 0x00000000  A8      : 0x00010000  A9      : 0x00000000  
> A10     : 0x00000001  A11     : 0x400811b8  A12     : 0x00000001  A13     : 0x3ffbd700
> A14     : 0x00000000  A15     : 0x3ffc192c  SAR     : 0x00000020  EXCCAUSE: 0x0000001d  
> EXCVADDR: 0x0001001c  LBEG    : 0x40086278  LEND    : 0x40086283  LCOUNT  : 0xffffffff
> 
> 
> Backtrace:0x4008140b:0x3ffbeddc |<-CORRUPTED
> 
> 
> 
> 
> ELF file SHA256: 0000000000000000

I tried looking at the source code of the library and in "Thyristors.cpp" on line 330:

> _allThyristorsOnOff = Thyristor::allThyristorsOnOff;

When she commented, the error no longer appeared, however, the dimmer started to work only in toggle mode, where on a scale of 0 to 5 my lamp was off and for brightness between 5 and 255 it was on at maximum brightness.

What I could also notice is that the problem is being caused in the hw_timer, because for a test done on the Arduino Mega 2560 the library worked perfectly.

If I could, I would very much like this problem to be resolved. At the moment, I will have to opt for using the Mega 2560 for my project.

99Arrzel, 2023-04-01:
@daliangowert Same problem with an esp32 and a RobotDyn Dimmer 2CH

fabianoriccardi, 2023-04-04:
May you try to replace line 115 in thyristor.cpp with:

    static bool DRAM_ATTR _allThyristorsOnOff = true;

? Basically DRAM_ATTR should force the variable to stay in RAM, so it should avoid the crash.

harsh0508, 2023-05-08:
Any updates ? 
I am getting the same problem

`
abort() was called at PC 0x40085f6f on core 1

Backtrace: 0x40083a4d:0x3ffbec8c |<-CORRUPTED

ELF file SHA256: 9b6eb8af02771d55
`

daliangowert, 2023-05-08:
hey! I'm without the RobotDyn 4ch module to test what @fabianoriccardi suggested. Did you try those changes he suggested?

doh38, 2023-05-18:
I'm on Liligo T3 (esp32 S3) and it behaves the same exact way. Commenting the line 330 out does the same, worked as a toggle (here at 13 ^^) 
so yes, probably a hw_timer issue.
Bit sad, this library sounded freaking promising :)

fabianoriccardi, 2023-11-01:
Closing since this issue was addressed in #41. A new version will be released soon.

closed 2023-11-01 by fabianoriccardi