fabianoriccardi/dimmable-light #30: Crash esp32 PlatfotmIO

dancheeek92, 2022-08-14:
Hello, thanks for the quality work! The problem is that at the moment the library does not work in PlatformIO, interrupt zc (pin 13) causes a crash. This issue has been asked before, but still no solution. You will do a great good deed if you deal with this problem. Your library is the only one that can provide an acceptable level of flicker.

D1ver, 2022-08-15:
@fabianoriccardi Can you help to us? In other issue you wrote, that problem is in new arduino core. May be you can update a library?

fabianoriccardi, 2022-08-19:
Hi, I will try to revamp the library to the newer version of Arduino cores and platformio. 

@dancheeek92 Which board and Arduino Core version are you using?

dancheeek92, 2022-08-21:
@fabianoriccardi esp32 wroom, PlatformIO Core 6.1.4 (2022-08-12). The problem was on previous versions as well.

fabianoriccardi, 2022-08-21:
May you tell me the ESP32-Arduino core version?

>  interrupt zc (pin 13) causes a crash. 

Is only pin 13 causing the crash or is it always crashing whatever the pin?

fabianoriccardi, 2022-08-22:
I haven't never heard about this board, and even if it based on ESP32, it seems the company has developed their own framework. Is it open source? Does it use the well-known [ESP32-Arduino](https://github.com/espressif/arduino-esp32)?

nsummy, 2022-09-06:
@dancheeek92 According to this:  https://www.briki.org/download/Pin_description_MBC-WB.pdf   Pins 13 and 27 are reserved for the onboard samd21...

fabianoriccardi, 2023-11-01:
Closing since this issue was addressed in #41. A new version will be released soon.

closed 2023-11-01 by fabianoriccardi