fabianoriccardi/dimmable-light #32: Code changes to allow max brightness of 4095?

nsummy, 2022-09-09:
The samd21 chip has the ability to read and write at 12 bits.  I realize such a high resolution is overkill for something like that but I am working on a project where I want to output pwm at 12 bits along with using this dimmer module.  It looks like I would need to make changes in [dimmable_light.h](https://github.com/fabianoriccardi/dimmable-light/blob/c935351fc173e1a1ec6c1bb3586f58e24bdccee1/src/dimmable_light.h) or [thyristor.cpp](https://github.com/fabianoriccardi/dimmable-light/blob/c935351fc173e1a1ec6c1bb3586f58e24bdccee1/src/thyristor.cpp) but not sure exactly where.  Any input would be appreciated. thank you

fabianoriccardi, 2022-09-11:
Yes it is an overkill for the common usage of a dimmer... 

What resolution do you need? Do you need such resolution on the dimmer or you need to share that timer for other usage?

nsummy, 2022-09-15:
As far as I know, once I set the write resolution to 12, its global for all timers.  But yes, basically just needed to make the max brightness 4095.  Your answer worked.  I also had to change this:  `uint16_t newDelay = 8333 - (uint16_t)(((uint32_t)bri * 8333) / 4095);`

Thanks!

closed 2022-09-15 by nsummy