fabianoriccardi/dimmable-light #35: Brightness works like 60hz but it should be 50hz

pabloeorsi, 2023-03-18:
The brightness behaves strangely.
At a brightness of value 1, it already has considerable intensity.
At values close to 213, 214 decreases sharply again, leaving a very low brightness.
At 255 the brightness is fully turned on again.

It seems to be running at a frequency of 60hz, when it is set to run at 50hz. On another occasion it worked without problems, I don't know now what the problem could be.

I am using an ESP8266 Wemos mini.

I hope you can help me.

pabloeorsi, 2023-03-27:
Hello again!

The frequency error is discarded.
From what I can conclude, I have a defase when interpreting the zero crossing, so the delay begins to pass before the wave passes by zero.
This generates that the minimum value, 1 of 255 generates low intensity light when it should be off.
The same happens with 215 approx, the light takes zero value and begins to gradually increase to 255.

I have a Robotdyn type zero crossing detector.
I already made many physical changes, but to refine the zero crossing I must gather the resistance in 220VCA dissipating a lot of power and burning in some cases the 4N25.

Is there any configuration of the library that allows me to adjust this delay?

Thank you very much for sharing!!!

Greetings!

fabianoriccardi, 2023-03-27:
Robotdyn dimmers are not the best since the zero cross detection circuitry triggers too early... I had already observed this issue but what you describe see too strong... The following couple of constants in thyristors.cpp: 

> static const uint16_t startMargin = 200;
> static const uint16_t endMargin = 500;

are "cut-on" and "cut-off" thresholds to mitigate this issue. Try to increase them (e.g. 1000us) . Here a more detailed description: https://github.com/fabianoriccardi/dimmable-light/blob/c7803314f92708f895227ede46426f68ac7863e4/src/thyristor.cpp#L69

pabloeorsi, 2023-04-04:
Hello @fabianoriccardi!

Thank you very much for your answer and for this great bookstore!
Modifying the startMargin parameter solves the upper limit, since it keeps the thyristor on after the zero crossing.
The problem is that I can't solve for low dimmer values, where the lowest value, 1 of 255, generates a large shooting angle, going from on to off, without low regulation.
If a delay parameter could be introduced in the zero crossing interrupt, the problem would be solved, since it could be adjusted so that it detects the crossing as close to the real value.

Again, thank you very much and congratulations, it is a super complete library.

Regards!

pabloeorsi, 2023-04-08:
Hello!

After some tests I concluded that the problem is that I have a very wide zero crossing pulse, because of the previous resistance to the optocoupler.
The problem is that if I put the proper resistor it consumes a lot of power, heating up too much.
If a delay time could be introduced in the detection of the zero crossing, it would be solved, and it would help to reduce the power consumption in this previous resistance.
I will mention it in the Pull request section.

Many thanks for everything!!

closed 2023-04-08 by pabloeorsi