fabianoriccardi/dimmable-light #38: Does not compile with WiFiManager library on ESP8266

heartwerker, 2023-07-04:
Love this library a lot. now i meant to connect to wifi.

unfortunately when i am trying to combine it with the [tzapu/WiFiManager](https://github.com/tzapu/WiFiManager) library my (otherwise empty) project cant compile.

Using platformio, wemos d1 mini esp8266...

 _platformio.ini_:

```
[env:d1_mini]
platform = espressif8266@2.3.0
board = d1_mini
framework = arduino

lib_ldf_mode = deep+ # for wifimanager lib
lib_deps = 
    tzapu/WiFiManager@^0.16.0
	fabianoriccardi/Dimmable Light for Arduino@^1.5.0
```

_main.cpp_:

```
#include <Arduino.h>
void setup(){}
void loop(){}
```

_Error Output_:
```
Processing d1_mini (platform: espressif8266@2.3.0; board: d1_mini; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Removing unused dependencies...
Library Manager: Installing fabianoriccardi/Dimmable Light for Arduino @ ^1.5.0
Unpacking  [####################################]  100%
Library Manager: Dimmable Light for Arduino@1.5.0 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing ArduinoSTL
Unpacking  [####################################]  100%
Library Manager: ArduinoSTL@1.3.3 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (2.3.0) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 2.20601.191124 (2.6.1) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.20800.0 (2.8.0) 
 - toolchain-xtensa @ 2.40802.191122 (4.8.2)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- WiFiManager @ 0.16.0
|-- Dimmable Light for Arduino @ 1.5.0
Building in release mode
Compiling .pio/build/d1_mini/src/main.cpp.o
Generating LD script .pio/build/d1_mini/ld/local.eagle.app.v6.common.ld
Compiling .pio/build/d1_mini/libacb/ArduinoSTL/ArduinoSTL.cpp.o
Compiling .pio/build/d1_mini/libacb/ArduinoSTL/abi/abi.cpp.o
In file included from /Users/heartwerker/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Arduino.h:246:0,
                 from src/main.cpp:1:
/Users/heartwerker/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Updater.h:53:13: error: 'function' in namespace 'std' does not name a type
     typedef std::function<void(size_t, size_t)> THandlerFunction_Progress;
             ^
/Users/heartwerker/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Updater.h:121:19: error: expected ';' at end of member declaration
     UpdaterClass& onProgress(THandlerFunction_Progress fn);
                   ^
/Users/heartwerker/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Updater.h:121:56: error: expected ')' before 'fn'
     UpdaterClass& onProgress(THandlerFunction_Progress fn);
                                                        ^
/Users/heartwerker/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Updater.h:204:5: error: 'THandlerFunction_Progress' does not name a type
     THandlerFunction_Progress _progress_callback;
     ^
In file included from src/main.cpp:1:0:
/Users/heartwerker/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Arduino.h:251:12: error: 'std::isinf' has not been declared
 using std::isinf;
            ^
/Users/heartwerker/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Arduino.h:252:12: error: 'std::isnan' has not been declared
 using std::isnan;
Compiling .pio/build/d1_mini/libacb/ArduinoSTL/algorithm.cpp.o
            ^
Compiling .pio/build/d1_mini/libacb/ArduinoSTL/associative_base.cpp.o
Compiling .pio/build/d1_mini/libacb/ArduinoSTL/bitset.cpp.o
Compiling .pio/build/d1_mini/libacb/ArduinoSTL/char_traits.cpp.o
*** [.pio/build/d1_mini/src/main.cpp.o] Error 1
.pio/libdeps/d1_mini/ArduinoSTL/src/ArduinoSTL.cpp:79:2: warning: #warning "printf() will not be functional on this platform." [-Wcpp]
 #warning "printf() will not be functional on this platform."
  ^
In file included from /Users/heartwerker/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Arduino.h:246:0,
                 from .pio/libdeps/d1_mini/ArduinoSTL/src/ArduinoSTL.h:12,
                 from .pio/libdeps/d1_mini/ArduinoSTL/src/ArduinoSTL.cpp:1:
/Users/heartwerker/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Updater.h:53:13: error: 'function' in namespace 'std' does not name a type
     typedef std::function<void(size_t, size_t)> THandlerFunction_Progress;
             ^
/Users/heartwerker/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Updater.h:121:19: error: expected ';' at end of member declaration
     UpdaterClass& onProgress(THandlerFunction_Progress fn);
                   ^
/Users/heartwerker/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Updater.h:121:56: error: expected ')' before 'fn'
     UpdaterClass& onProgress(THandlerFunction_Progress fn);
                                                        ^
/Users/heartwerker/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Updater.h:204:5: error: 'THandlerFunction_Progress' does not name a type
     THandlerFunction_Progress _progress_callback;
     ^
In file included from .pio/libdeps/d1_mini/ArduinoSTL/src/ArduinoSTL.h:12:0,
                 from .pio/libdeps/d1_mini/ArduinoSTL/src/ArduinoSTL.cpp:1:
/Users/heartwerker/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Arduino.h:251:12: error: 'std::isinf' has not been declared
 using std::isinf;
            ^
/Users/heartwerker/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Arduino.h:252:12: error: 'std::isnan' has not been declared
 using std::isnan;
            ^
*** [.pio/build/d1_mini/libacb/ArduinoSTL/ArduinoSTL.cpp.o] Error 1
======================================================================================================================= [FAILED] Took 3.02 seconds =======================================================================================================================

 *  The terminal process "platformio 'run', '--environment', 'd1_mini'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

 any ideas??????

heartwerker, 2023-07-04:
There is also an Error when trying to use regular ESP8266 Wifi....

```
[env:d1_mini]
platform = espressif8266@2.3.0
board = d1_mini
framework = arduino

lib_deps = 
	fabianoriccardi/Dimmable Light for Arduino@^1.5.0

```

_main.cpp_:

```
#include <Arduino.h>
#include <ESP8266WiFi.h>
void setup(){}
void loop(){}
```

´´´
Processing d1_mini (platform: espressif8266@2.3.0; board: d1_mini; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing fabianoriccardi/Dimmable Light for Arduino @ ^1.5.0
Unpacking  [####################################]  100%
Library Manager: Dimmable Light for Arduino@1.5.0 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing ArduinoSTL
Unpacking  [####################################]  100%
Library Manager: ArduinoSTL@1.3.3 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (2.3.0) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 2.20601.191124 (2.6.1) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.20800.0 (2.8.0) 
 - toolchain-xtensa @ 2.40802.191122 (4.8.2)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Dimmable Light for Arduino @ 1.5.0
|-- ESP8266WiFi @ 1.0
Building in release mode
Compiling .pio/build/d1_mini/src/main.cpp.o
Generating LD script .pio/build/d1_mini/ld/local.eagle.app.v6.common.ld
Compiling .pio/build/d1_mini/libacb/ArduinoSTL/ArduinoSTL.cpp.o
Compiling .pio/build/d1_mini/libacb/ArduinoSTL/abi/abi.cpp.o
Compiling .pio/build/d1_mini/libacb/ArduinoSTL/algorithm.cpp.o
Compiling .pio/build/d1_mini/libacb/ArduinoSTL/associative_base.cpp.o
Compiling .pio/build/d1_mini/libacb/ArduinoSTL/bitset.cpp.o
In file included from /Users/davidspeith/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Arduino.h:246:0,
                 from src/main.cpp:1:
/Users/davidspeith/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Updater.h:53:13: error: 'function' in namespace 'std' does not name a type
     typedef std::function<void(size_t, size_t)> THandlerFunction_Progress;
             ^
/Users/davidspeith/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Updater.h:121:19: error: expected ';' at end of member declaration
     UpdaterClass& onProgress(THandlerFunction_Progress fn);
                   ^
/Users/davidspeith/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Updater.h:121:56: error: expected ')' before 'fn'
     UpdaterClass& onProgress(THandlerFunction_Progress fn);
                                                        ^
/Users/davidspeith/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Updater.h:204:5: error: 'THandlerFunction_Progress' does not name a type
     THandlerFunction_Progress _progress_callback;
     ^
[... 161 lines cut ...]
             ^
/Users/davidspeith/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Updater.h:121:19: error: expected ';' at end of member declaration
     UpdaterClass& onProgress(THandlerFunction_Progress fn);
                   ^
/Users/davidspeith/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Updater.h:121:56: error: expected ')' before 'fn'
     UpdaterClass& onProgress(THandlerFunction_Progress fn);
                                                        ^
/Users/davidspeith/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Updater.h:204:5: error: 'THandlerFunction_Progress' does not name a type
     THandlerFunction_Progress _progress_callback;
     ^
In file included from .pio/libdeps/d1_mini/ArduinoSTL/src/ArduinoSTL.h:12:0,
                 from .pio/libdeps/d1_mini/ArduinoSTL/src/ArduinoSTL.cpp:1:
/Users/davidspeith/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Arduino.h:251:12: error: 'std::isinf' has not been declared
 using std::isinf;
            ^
/Users/davidspeith/.platformio/packages/framework-arduinoespressif8266@2.20601.191124/cores/esp8266/Arduino.h:252:12: error: 'std::isnan' has not been declared
 using std::isnan;
            ^
*** [.pio/build/d1_mini/libacb/ArduinoSTL/ArduinoSTL.cpp.o] Error 1
================================================================================================================= [FAILED] Took 1.95 seconds =================================================================================================================

 *  The terminal process "platformio 'run', '--environment', 'd1_mini'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

´´´

fabianoriccardi, 2023-11-12:
You can fix it by adding `lib_compat_mode = strict` in `platformio.ini`.

closed 2023-11-12 by fabianoriccardi

heartwerker, 2023-11-13:
Great, thank you !