fabianoriccardi/dimmable-light #42: MQTT error

Prithvikd, 2023-11-01:
I am trying to implement mqtt and i'm are facing a problem as there are two interrupts mqtt and a code for dimmable lights. these both interrupts are not working together, please help me out

fabianoriccardi, 2023-11-01:
Which version of ESP32 Core are you using?

Prithvikd, 2023-11-01:
> Which version of ESP32 Core are you using?

ESP 32 WROOM 32D
Xtensa dual core 

Platformio: espressif32 3.5.0

fabianoriccardi, 2023-11-01:
Espressif32 3.5.0 corresponds to Arduino Esp32 v1.0.6.

I have tested with Arduino IDE 2 the following sketch and it compiles:

``` C 
#include <WiFi.h>
#include <MQTT.h>

#include <dimmable_light.h>

DimmableLight light(3);

WiFiClient net;
MQTTClient client;

void setup() {
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

Can you post the minimal sketch that throws the error?

Prithvikd, 2023-11-02:
> Espressif32 3.5.0 corresponds to Arduino Esp32 v1.0.6.
>
> I have tested with Arduino IDE 2 the following sketch and it compiles:
>
> ```c
> #include <WiFi.h>
> #include <MQTT.h>
>
> #include <dimmable_light.h> [...]

#include <WiFi.h>
#include <MQTT.h>

#include <dimmable_light.h>

DimmableLight light(3);

WiFiClient net;
MQTTClient client;

void incomingMessageHandler(String &topic, String &payload){

//connect to AWS 
}

void setup() {
  Serial.begin(115200);
//wifi connection with mqtt client connection to aws with the certificates
mqtt_client_incommingmessagehandler ();
}

void loop() {

mqtt_client . loop();
}

fabianoriccardi, 2023-11-02:
Please post a correct sketch. For sure 'mqtt_client_incommingmessagehandler' is not declared function

Prithvikd, 2023-11-03:
> Please post a correct sketch. For sure 'mqtt_client_incommingmessagehandler' is not declared function

#include<dimmable_light.h>
#include <WiFi.h>
#include <MQTT.h>
#include <ArduinoJson.h>
#include <WiFiClient.h>

DimmableLight light(3);

WiFiClient wifi_client;
MQTTClient mqtt_client;
const char *AWS_ENDPOINT = "xxxxxxxxxxxxxxx-ats.iot.us-east-1.amazonaws.com";

void incomingMessageHandler(String &topic, String &payload) {
  Serial.println("incoming: " + topic + " - " + payload);
  DynamicJsonDocument doc(1024);
  deserializeJson(doc, payload);
  if (doc.containsKey("brightness")) {
    light.setBrightness(doc["brightness"]);
  }
}
connectWifi()
{
  //connect to wifi
}
AWSconnect()
{
  //connect to AWS
  mqtt_client.begin(AWS_ENDPOINT, wifi_client);
  mqtt_client.onMessage(incomingMessageHandler);
}
void setup()
{
  Serial.begin(115200);
  connectWifi();
  AWSconnect();
  Serial.print("Initializing DimmableLight library... ");
  DimmableLight::setSyncPin(syncPin);
  DimmableLight::begin();
}

void loop()
{
  mqtt_client.loop();
}

fabianoriccardi, 2023-11-03:
It seems you don't want the help: the code doesn't compile! I had solved 3 trivial issues that prevents compilation (missing return values of functions and missing variable declaration) and it works seamlessly. 

Post the [MCVE](https://stackoverflow.com/help/minimal-reproducible-example). You say that you are using platformio, so you should also post the platformio.ini file.

reddy9698, 2023-11-06:
Hi, I encountered the same problem.

This is the basic code:

`#include <dimmable_light.h>
#include<Arduino.h>
#include<WiFi.h>
#include<ArduinoJson.h>
#include<MQTTClient.h>
#include<WiFiClientSecure.h>

#define THINGNAME "dimlight"

const char*SSID = "SSID";
const char*password = "PASS";
const char endpoint[] = "xxxxxxxxxx-ats.iot.ap-south-1.amazonaws.com";

const int syncPin = 5;
// const int thyristorPin = 19;

DimmableLight light1(12);
DimmableLight light2(14);
DimmableLight light3(27);
DimmableLight light4(26);
DimmableLight light5(13);

// Delay between brightness increments, in milliseconds
const int period = 50;

#define sub_topic "$aws/things/dimlight/shadow/update/delta"
#define pub_topic "$aws/things/dimlight/shadow/update"

WiFiClientSecure wifi_client=WiFiClientSecure();
MQTTClient mqtt_client=MQTTClient(2048);

static const char ROOT_CA[] PROGMEM= R"EOF(
-----BEGIN CERTIFICATE-----
MIIDQTCCAimgAwIBAgITBmyfz5m/jAo54vB4ikPmljZbyjANBgkqhkiG9w0BAQsF
ADA5MQswCQYDVQQGEwJVUzEPMA0GA1UEChMGQW1hem9uMRkwFwYDVQQDExBBbWF6
b24gUm9vdCBDQSAxMB4XDTE1MDUyNjAwMDAwMFoXDTM4MDExNzAwMDAwMFowOTEL
MAkGA1UEBhMCVVMxDzANBgNVBAoTBkFtYXpvbjEZMBcGA1UEAxMQQW1hem9uIFJv
b3QgQ0EgMTCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBALJ4gHHKeNXj
ca9HgFB0fW7Y14h29Jlo91ghYPl0hAEvrAIthtOgQ3pOsqTQNroBvo3bSMgHFzZM
9O6II8c+6zf1tRn4SWiw3te5djgdYZ6k/oI2peVKVuRF4fn9tBb6dNqcmzU5L/qw
IFAGbHrQgLKm+a/sRxmPUDgH3KKHOVj4utWp+UhnMJbulHheb4mjUcAwhmahRWa6
VOujw5H5SNz/0egwLX0tdHA114gk957EWW67c4cX8jJGKLhD+rcdqsq08p8kDi1L
93FcXmn/6pUCyziKrlA4b9v7LWIbxcceVOF34GfID5yHI9Y/QCB/IIDEgEw+OyQm
jgSubJrIqg0CAwEAAaNCMEAwDwYDVR0TAQH/BAUwAwEB/zAOBgNVHQ8BAf8EBAMC
AYYwHQYDVR0OBBYEFIQYzIU07LwMlJQuCFmcx7IQTgoIMA0GCSqGSIb3DQEBCwUA
A4IBAQCY8jdaQZChGsV2USggNiMOruYou6r4lK5IpDB/G/wkjUu0yKGX9rbxenDI
U5PMCCjjmCXPI6T53iHTfIUJrU6adTrCC2qJeHZERxhlbI1Bjjt/msv0tadQ1wUs
N+gDS63pYaACbvXy8MWy7Vu33PqUXHeeE6V/Uq2V8viTO96LXFvKWlJbYK8U90vv
o/ufQJVtMVT8QtPHRh8jrdkPSHCa2XV4cdFyQzR1bldZwgJcJmApzyMZFo6IQ6XU
5MsI+yMRQ+hDKXJioaldXgjUkK642M4UwtBV8ob2xJNDd2ZhwLnoQdeXeGADbkpy
rqXRfboQnoZsG4q5WTP468SQvvG5
-----END CERTIFICATE-----
)EOF";

static const char PRIVATE_KEY[] PROGMEM = R"KEY(
-----BEGIN RSA PRIVATE KEY-----
// your key here
-----END RSA PRIVATE KEY-----

 )KEY";

static const char  CERTFICATE_KEY [] PROGMEM = R"KEY(
-----BEGIN CERTIFICATE-----
// your key here
-----END CERTIFICATE-----

)KEY";

void IncomingMessageHandler (String &topic, String&payload)
{
    Serial.println("Message received"); 
    Serial.println("Topic" +String(topic));
    Serial.println("payload" + String(payload));
}

void setup()
{
  Serial.begin(115200);

  while (!Serial)
    ;
  Serial.println();
  WiFi.mode(WIFI_STA);
  WiFi.begin (SSID,password); 
  while(WiFi.status()!=WL_CONNECTED)
  {
      Serial.print(".");
      delay(50);

  }
  Serial.println("WiFi Connected");
  Serial.println(WiFi.localIP());  
  DimmableLight::setSyncPin(syncPin);
  DimmableLight::begin();
  // Serial.println("Done!");
  wifi_client.setCACert(ROOT_CA);
    wifi_client.setCertificate(CERTFICATE_KEY);
    wifi_client.setPrivateKey(PRIVATE_KEY);
    mqtt_client.begin(endpoint,8883,wifi_client);
    mqtt_client.onMessage(IncomingMessageHandler);
    Serial.println("connecting to AWS");
    mqtt_client.connect(THINGNAME);
    while(!mqtt_client.connect(THINGNAME));
    {
        Serial.print("*");
        delay(50);
    }
    Serial.print("connected To AWS");
    if(!mqtt_client.connected())
    {
        Serial.println("Connection Timeout");
    }
    mqtt_client.subscribe(sub_topic);
}

void loop()
{
  mqtt_client.loop();

}`

and this is the platformio.ini file:
`[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps = 
    bblanchon/ArduinoJson@^6.21.3
    256dpi/MQTT@^2.5.0
    fabianoriccardi/Dimmable Light for Arduino@^1.5.0
monitor_speed = 115200`

The libraries MQTT and dimmable light work very good separately. The problem arises when I include the MQTT part with dimmable light.

fabianoriccardi, 2023-11-07:
I post here the version that really gives the issue(😅): 

``` Cpp
#include <WiFi.h>
#include <MQTTClient.h>
#include <WiFiClientSecure.h>
#include <dimmable_light.h>

#define THINGNAME "dimlight"

const char SSID[] = "SSID";
const char password[] = "PASS";
const char endpoint[] = "xxxxxxxxxx-ats.iot.ap-south-1.amazonaws.com";

const int syncPin = 5;
// const int thyristorPin = 19;

DimmableLight light1(12);

// Delay between brightness increments, in milliseconds
const int period = 50;

#define sub_topic "$aws/things/dimlight/shadow/update/delta"
#define pub_topic "$aws/things/dimlight/shadow/update"

WiFiClientSecure wifi_client = WiFiClientSecure();
MQTTClient mqtt_client = MQTTClient(2048);

static const char ROOT_CA[] PROGMEM = R"EOF(
-----BEGIN CERTIFICATE-----
MIIDQTCCAimgAwIBAgITBmyfz5m/jAo54vB4ikPmljZbyjANBgkqhkiG9w0BAQsF
ADA5MQswCQYDVQQGEwJVUzEPMA0GA1UEChMGQW1hem9uMRkwFwYDVQQDExBBbWF6
b24gUm9vdCBDQSAxMB4XDTE1MDUyNjAwMDAwMFoXDTM4MDExNzAwMDAwMFowOTEL
MAkGA1UEBhMCVVMxDzANBgNVBAoTBkFtYXpvbjEZMBcGA1UEAxMQQW1hem9uIFJv
b3QgQ0EgMTCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBALJ4gHHKeNXj
ca9HgFB0fW7Y14h29Jlo91ghYPl0hAEvrAIthtOgQ3pOsqTQNroBvo3bSMgHFzZM
9O6II8c+6zf1tRn4SWiw3te5djgdYZ6k/oI2peVKVuRF4fn9tBb6dNqcmzU5L/qw
IFAGbHrQgLKm+a/sRxmPUDgH3KKHOVj4utWp+UhnMJbulHheb4mjUcAwhmahRWa6
VOujw5H5SNz/0egwLX0tdHA114gk957EWW67c4cX8jJGKLhD+rcdqsq08p8kDi1L
93FcXmn/6pUCyziKrlA4b9v7LWIbxcceVOF34GfID5yHI9Y/QCB/IIDEgEw+OyQm
jgSubJrIqg0CAwEAAaNCMEAwDwYDVR0TAQH/BAUwAwEB/zAOBgNVHQ8BAf8EBAMC
AYYwHQYDVR0OBBYEFIQYzIU07LwMlJQuCFmcx7IQTgoIMA0GCSqGSIb3DQEBCwUA
A4IBAQCY8jdaQZChGsV2USggNiMOruYou6r4lK5IpDB/G/wkjUu0yKGX9rbxenDI
U5PMCCjjmCXPI6T53iHTfIUJrU6adTrCC2qJeHZERxhlbI1Bjjt/msv0tadQ1wUs
N+gDS63pYaACbvXy8MWy7Vu33PqUXHeeE6V/Uq2V8viTO96LXFvKWlJbYK8U90vv
o/ufQJVtMVT8QtPHRh8jrdkPSHCa2XV4cdFyQzR1bldZwgJcJmApzyMZFo6IQ6XU
5MsI+yMRQ+hDKXJioaldXgjUkK642M4UwtBV8ob2xJNDd2ZhwLnoQdeXeGADbkpy
rqXRfboQnoZsG4q5WTP468SQvvG5
-----END CERTIFICATE-----
)EOF";

static const char PRIVATE_KEY[] PROGMEM = R"KEY(
-----BEGIN RSA PRIVATE KEY-----
// your key here
-----END RSA PRIVATE KEY-----

)KEY";

static const char CERTFICATE_KEY[] PROGMEM = R"KEY(
-----BEGIN CERTIFICATE-----
// your key here
-----END CERTIFICATE-----

)KEY";

void IncomingMessageHandler(String &topic, String &payload)
{
  Serial.println("Message received");
  Serial.println("Topic" + String(topic));
  Serial.println("payload" + String(payload));
}

void setup()
{
  Serial.begin(115200);

  while (!Serial)
    ;
  Serial.println();
  WiFi.mode(WIFI_STA);
  WiFi.begin(SSID, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(50);
  }
  Serial.println("WiFi Connected");
  Serial.println(WiFi.localIP());
  DimmableLight::setSyncPin(syncPin);
  DimmableLight::begin();
  // Serial.println("Done!");
  wifi_client.setCACert(ROOT_CA);
  wifi_client.setCertificate(CERTFICATE_KEY);
  wifi_client.setPrivateKey(PRIVATE_KEY);
  mqtt_client.begin(endpoint, 8883, wifi_client);
  mqtt_client.onMessage(IncomingMessageHandler);
  Serial.println("connecting to AWS");
  mqtt_client.connect(THINGNAME);
  while (!mqtt_client.connect(THINGNAME))
    ;
  {
    Serial.print("*");
    delay(50);
  }
  Serial.print("connected To AWS");
  if (!mqtt_client.connected())
  {
    Serial.println("Connection Timeout");
  }
  mqtt_client.subscribe(sub_topic);
}

void loop()
{
  mqtt_client.loop();
}
```

fabianoriccardi, 2023-11-07:
The problem is simple: PIO includes ArduinoSTL library even if not necessary on esp32 platform. In library.json there is the configuration that explicitly tells that ArduinoSTL is only needed for Arduino AVR, but PIO, by default behavior, includes it in every platformio.

To induce PIO to consider the specific platform, you have to add the following line in `platformio.ini`:

    lib_compat_mode = strict

[docs](lib_compat_mode = strict)

fabianoriccardi, 2023-11-11:
Add a note in readme in  https://github.com/fabianoriccardi/dimmable-light/commit/b40c7c4565c2a3aaabc35ca450019fc4cc25d052.

I'm closing since I didn't find any better solution. If you still have problems, let me know.

closed 2023-11-11 by fabianoriccardi

reddy9698, 2023-11-11:
Hi, your solution works fine for now. Thank you for the fast response.