fabianoriccardi/dimmable-light #44: ESP32 - IRAM interrupts

florentbr, 2023-11-05:
I noticed that the current implementation uses non IRAM functions in the ESP32 interrupts :

https://github.com/fabianoriccardi/dimmable-light/blob/21edb0dde6cf61e4c866b704db18c99f4eaae86f/src/hw_timer_esp32.cpp#L39-L56

All the interrupt code needs to be placed into IRAM to avoid long latency or crash when the flash is locked.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/intr_alloc.html#iram-safe-interrupt-handlers

Either replace the timer functions with specific isr timer functions from `timer.h`:

```c++
timer_group_set_alarm_value_in_isr(...)
timer_group_set_counter_enable_in_isr(...)
timer_group_enable_alarm_in_isr(...)
```

or with inlined low level functions from `timer_ll.h` :

```c++
#define TIMER_NUM  ( 2 )
#define TIMER_HW   ( TIMER_LL_GET_HW(TIMER_NUM / SOC_TIMER_GROUPS) )
#define TIMER_ID   ( TIMER_NUM % SOC_TIMER_GROUP_TIMERS_PER_GROUP )

timer_ll_set_counter_value(TIMER_HW, TIMER_ID, 0);
timer_ll_set_alarm_value(TIMER_HW, TIMER_ID, 5000);
timer_ll_set_alarm_enable(TIMER_HW, TIMER_ID, true);
```

fabianoriccardi, 2023-11-05:
Thanks for the observation. I had replaced the IRAM_ATTR with `ARDUINO_ISR_IRAM` [Docs](https://github.com/espressif/arduino-esp32/blob/6d9ebea5008be540aaea5ad3cffc6ceb583692e3/Kconfig.projbuild#L150) in the latest release because it doesn't make sense placing my library functions in IRAM if the Arduino HAL (which I rely on) is not there. Moreover, in the previous releases, from what I can understand, IRAM_ATTR wasn't providing a real benefit because I was registering interrupts without `ESP_INTR_FLAG_IRAM`.

Given the current implementation, I except possible "long latency" but no crashes. I don't know how much this latency can be. 

As you point out, I may replace all Arduino HAL calls with "native" low-level calls about timers and GPIOs, but it will require some effort. I can consider it.

florentbr, 2023-11-06:
Using IRAM_ATTR or ARDUINO_ISR_IRAM on an interrupt is pretty much useless if the interrupt itself is calling even a single non IRAM function. The ESP_INTR_FLAG_IRAM is there to place the interrupt handler in IRAM as well.

The doc says the ARDUINO_ISR_IRAM flag makes some HAL function, like digitalRead/Write and more to be loaded into IRAM. Looking at the Arduino sources, [`digitalWrite`](https://github.com/espressif/arduino-esp32/blob/6d9ebea5008be540aaea5ad3cffc6ceb583692e3/cores/esp32/esp32-hal-gpio.c#L154) is indeed tagged as IRAM but somehow it rely on `gpio_set_level` which is not. There's a closed issue about it on github: https://github.com/espressif/esp-idf/issues/2408 .

What's disturbing is that the hal functions `timerWrite`, `timerAlarmWrite`, `timerAlarmEnable`, `timerStart` from Arduino (`esp32-hal-timer.h`)[https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-timer.c] are not tagged IRAM but the ones from IDF (`hal/timer_hal.h`)[https://github.com/espressif/esp-idf/blob/v4.4.5/components/hal/include/hal/timer_hal.h] are.

All the code from Espressif are using interrupts with code executed exclusively from IRAM.
Since this library has no control over the rest of the code, it's probably best to implement the interrupts in IRAM as recommended. From Espressif:

> It can also be useful to keep an interrupt handler in IRAM if it is called very frequently, to avoid flash cache misses.

Some discussions on the subject:
https://github.com/espressif/arduino-esp32/issues/3697
https://esp32.com/viewtopic.php?t=4978

fabianoriccardi, 2023-11-06:
The routine will be called 100 times/second if AC@50Hz, so it will stay in IRAM very probably and you won't have problems. The same for the other ISRs.

If your firmware performs a lot of flash operations will have sporadic huge latencies, and this is bad. 
As you point out, there is not consistency in HAL IRAM_ATTR usage, so, the only way to completely solve this issue is to use registers or the lowest-level functions that are in IRAM. 

About [timer_ll.h](https://github.com/espressif/esp-idf/blob/v4.4.5/components/hal/esp32/include/hal/timer_ll.h) [gpio_ll.h](https://github.com/espressif/esp-idf/blob/v4.4.5/components/hal/esp32/include/hal/gpio_ll.h)
Those functions are inline, not explicitly in IRAM but as consequence they should be embedded in IRAM as the caller. Same for *gpio_ll_set_level*.

Hence, there is a change to code an optimized version which stays in IRAM despite the Arduino config.

florentbr, 2023-11-06:
> About [timer_ll.h](https://github.com/espressif/esp-idf/blob/v4.4.5/components/hal/esp32/include/hal/timer_ll.h) [gpio_ll.h](https://github.com/espressif/esp-idf/blob/v4.4.5/components/hal/esp32/include/hal/gpio_ll.h)
Those functions are inline, not explicitly in IRAM but as consequence they should be embedded in IRAM as the caller. Same for gpio_ll_set_level.

You're right, sorry i meant to say that one inherits the IRAM attribute while the other (Arduino) doesn't.

fabianoriccardi, 2023-11-09:
Just for clarity, the issue #23 was related to your concerns, I will investigate it.

mathieucarbou, 2024-09-27:
FYI I have extracted a ZCD I am using for [YaSolR](https://yasolr.carbou.me/) (solar diverter) to https://github.com/mathieucarbou/MycilaPulseAnalyzer

I've managed to recreated some inlined versions of gptimer calls: they are isolated in a header. (`inlined_gptimer.h`)

Thanks to that I am able to compile the examples with 

```
  -D CONFIG_ARDUINO_ISR_IRAM=1
  -D CONFIG_GPTIMER_ISR_HANDLER_IN_IRAM=1
  -D CONFIG_GPTIMER_CTRL_FUNC_IN_IRAM=1
  -D CONFIG_GPTIMER_ISR_IRAM_SAFE=1
  -D CONFIG_GPIO_CTRL_FUNC_IN_IRAM=1
```

And have all the ISR code loaded into IRAM.

The 2 examples are doing concurrent flash (nvs) operations to prove that there is no crash anymore (before, a crash would happen immediately or after a few seconds).

This is not as optimised as the code @florentbr has done (triac_timer.cpp), but it it far less complex to use eventually in a project like dimmable light.

Also, using gpio_ll instead of arduino GPUI functions is quite svraitworward.

Note: my library only work with Arduino >= 3 and esp32 boards.

mathieucarbou, 2025-11-26:
> Thanks for the observation. I had replaced the IRAM_ATTR with `ARDUINO_ISR_IRAM` [...]

Hello experts,

So that's finally what I did in my dimming library: https://github.com/mathieucarbou/MycilaDimmer

I have some test examples running some concurrent flash operations while dimming to verify that everything is working and stays in IRAM.
- If CONFIG_ARDUINO_ISR_IRAM=1 => dimming is nearly not impacted
- CONFIG_ARDUINO_ISR_IRAM=0 => dimming is heavily impacted

Overall it works fine but ,with heavy flash operations, the dimming is still a little impacted (some flickering). But this is minor and does not crash.

I chose to rewrite the non-inlined gptimer functions, which I call directly instead of going through the arduino api. I did that because I wanted to stay as close as possible to the arduino / esp-idf abstract layers on top of timer_ll.h gpio_ll.h which are more complex to use.

transition is also smoother than with dimmable light.

I used Florent's idea for the power LUT which I have increased to 200 values.