fabianoriccardi/dimmable-light #46: `DimmableLight::getDetectedFrequency()` returns wrong value on Robodyn AC Dimmer 24A

mathieucarbou, 2023-12-20:
- v1.6.0
- Robodyn AC Dimmer 24A
- Inspiration from example: `8_set_frequency_automatically.ino`
- Only using ZC pin (control pin is not used)
- France, 50 Hz, 230V

My code: 

```c
DimmableLight::setSyncPin(_pin); // ZC pin of Robodyn
DimmableLight::begin();
```

Calling after, each 5 sec: `DimmableLight::getDetectedFrequency()`

Returned value is 122.

mathieucarbou, 2023-12-20:
Hello again,

I took some measurements this evening (Owon HDS2202S) and it turns out the Robodyn ZCD pulse is really short: still 5V, but really short, like 0.5 ms, compared to the 1.5 ms duration of the pulse of the dedicated ZCD.

I am wondering if there are some issues detecting that in the code ?

[image]

[image]

fabianoriccardi, 2023-12-21:
Hi Mathieu, which mcu are you using?

mathieucarbou, 2023-12-21:
> Hi Mathieu, which mcu are you using?

ESP32 NodeMCU devkit

This one: https://www.az-delivery.de/en/products/esp32-developmentboard

I've read the Wiki and learned about some ESP32 issues to detect short pulses, I am trying to activate filters now...

mathieucarbou, 2023-12-21:
I added `-D FILTER_INT_PERIOD` and everything works fine with both the Robodyn and the other ZCD module.

Thanks a lot for this library by the way: far better written than most others, and goes deeper in the details 👍

closed 2023-12-21 by mathieucarbou

fabianoriccardi, 2023-12-21:
> ESP32 NodeMCU devkit

I was supposing it, ESP32 gave quite few troubles, especially when combined with the RobotDyn dimmer. 😅 Glad that you quickly solved it!

> Thanks a lot for this library by the way: far better written than most others, and goes deeper in the details 👍

Many thanks!

mathieucarbou, 2023-12-21:
... And feel free to use the images above if you need, to illustrate your wiki page or put another ZCD module in ref ;-)

fabianoriccardi, 2023-12-22:
That's a good point, in the next days I hope to continue the work with this lib.

fabianoriccardi, 2023-12-23:
Add the reference and the images to the wiki page, thanks again!