fabianoriccardi/dimmable-light #48: Arduino Nano (AVR): works very well. Arduino Nano 33 iot : flickering.

spacejunkies, 2024-01-07:
**Describe the bug**
When I use the Arduino Nano (AVR), the dimmer works very well. When I use the Arduino Nano 33 iot (SAMD), there is flickering. 

**Hardware and software used**

- Library version: [1.6.0]
- Arduino core: [Arduino SAMD v1.8.13]
- Enviroment/IDE: [Arduino IDE]
- Dimmer board: [RobotDYN 2-channel]
- Board: [Arduino Nano 33 IoT]

fabianoriccardi, 2024-01-07:
Hi, had you tried to adjust (and increase) the constants such as startMargin, endMargin, and gateTurnOffTime at the begin of `thyristor.c`?

spacejunkies, 2024-01-08:
Hello, thank you for your quick reply. I will try different values for startMargin, endMargin, and gateTurnOffTime. Would you recommend specific ranges of value to try for those parameters?
In the meantime, here is the link to the RobotDyn module I am using: 
https://www.aliexpress.com/item/32972336897.html?pdp_npi=4%40pre%21CAD%21C%24%2011.17%21%21C%24%2011.17%21%218.18%217.93%21%402101f08717046829714527105ed6da%2112000016908124674%21sh01%21CA%21949636932%21&spm=a2g0o.store_pc_home.productList_2006959807184.32972336897

fabianoriccardi, 2024-01-08:
The 2 modules have slightly different electronics, but it shouldn't be a problem...

An easy trial may be respectively 800, 800, 700us, and then tune them by steps of 100us.

spacejunkies, 2024-01-10:
You are correct: the library works with Arduino Nano 33 iot. However, under certain conditions, flickering can be observed. In summary, there is a design problem with the Arduino Nano 33 iot board design that causes an issue then the board is powered with Vin. Thank you for your support - and let me explain (it could help another user having issue with your dimmer library.)

I found out the problem by accident when I disconnected the board from the 12V Power Supply that was connected to Vin, and powered it through the USB port. I observed that the flickering was less intense. Curious, I then used a external variable power supply that I connected to Vin. As I increased the voltage, the flickering became, indeed, more intense. I then researched to see if anyone else had problems when Vin is used. And I found many users reporting issues. Example: 
https://forum.arduino.cc/t/extreme-noisy-board-voltage-3-3v-on-nano-33-iot/623016

The conclusion from all those threads: the power regulator on the Arduino Nano 33 iot was not designed in correctly, and generates a very noisy 3.3V power line. It creates performance issues with some pins. So I set my external power supply to 3.3V and connected it to the 3.3V pin (bypassing the power regulator altogether). And the flickering pretty much disappeared!

The solution for my prototype: I will use an external power regulator, that will step down the 12V into 3.3V and I will connect it directly to the Arduino Nano 33iot 3.3V pin. 
https://www.amazon.ca/gp/product/B08214MRSW/ref=ox_sc_act_title_2?smid=A30Y6WWS77DGEW&psc=1

fabianoriccardi, 2024-01-21:
Thanks for reporting the link about the Arduino IoT 33 issue, for sure it may help other developers!