fabianoriccardi/dimmable-light #50: Full-wave mode implementation

mathieucarbou, 2024-01-21:
I would be interested by a Full-wave mode implementation like Circuitar, in order to provide another algorithm limiting harmonic effects.

I will try to implement it.

fabianoriccardi, 2024-01-21:
You are not the first that ask for it... I had thought to this feature in the past, but I have never found the time to implement it. If you want to implement it, you are welcome.

fabianoriccardi, 2024-01-21:
Before opening the PR, we should choose the APIs to configure full-wave mode.

mathieucarbou, 2024-01-21:
> Before opening the PR, we should choose the APIs to configure full-wave mode.

Of course! Were you thinking of something in specific ? 

I didn't start yet, but from what I quickly saw, since there are 2 dimmer implementations, I was thinking about adding a third one. 

I don't like the name "count mode": I prefer burst fire or maybe train wave modulation (?).

Also, there might be several ways to simplement that (project page website: https://yasolr.carbou.me/overview#burst-fire-modulation). 

I was planning to start first with a simple version of burst mode, using only a complete wave, except that I wouldn't do the 50% the same way but instead have something like: 

`+ - _ _ + - _ _ + - _ _ + - _ _`

- `+` == complete top half of the sinusoidal curve (positive side)
- `-` == complete bottom half of the sinusoidal curve (negative side)
- `_` == triac stays opened 

The equivalent 50% could be done like that also:

`+ - + - + - + - _ _ _ _ _ _ _ _`

I think the first one is better for a resistive load since it maintains the heat better, and when using the algorithm in the case of a solar router, it is best to have close followings of import and export to avoid excessive consumption recorded by the utility meter.

In a second time, I would like to implement a better algorithm where we control but per half wave to optimise sequence of import and export in order to better trick the utility meter.

At 50 Hz, we have 50 periods of 20 ms, each one being a complete wave of 2 half waves, which means we match the nominal consumption of the resistance within 20ms. Considering half-waves, we have 100 semi-periods of 10ms each, so I'd like to squeeze as many sequences of `+ _` or `- _` in order, within 20 ms to let some import (consumption) pass and some export pass (solar excess), in order for the grid utility meter to see 0 at the end.

So 50% could be done with a sequence of: `+ _ + _ + _ + _` in alternance with a sequence of `- _ - _ - _ - _` (number to be determined), basically in order to avoid introducing a DC component for a long period because cutting the half wave is similar to a current rectifier.

Anyway... Al that is just draft in my head atm, so nothing written yet, but still, this is something I have to d oto complete the project.

mathieucarbou, 2025-11-26:
fyi I am closing this. 
I have since developed a far better dimming library with a superior control over fading and steps, and a correct handling of ISR.
https://github.com/mathieucarbou/MycilaDimmer
I will also add support for cycle stealing, either wit ha deterministic pattern or by using a stochastic approach with random levels.

closed 2025-11-26 by mathieucarbou