fabianoriccardi/dimmable-light #66: Error compilation with the library Arduino core for the ESP32 version 3.0.

pepuma, 2024-07-09:
**Describe the bug**
A clear description of what the bug is.

**Hardware and software used**

- Library version: [eg: 1.5.1]
- Arduino core: [eg: SAMD v1.8.13]
- Enviroment/IDE: [Arduino IDE or PlatformIO]
- Dimmer board: [eg: Krida's 8 channel dimmers]
- Board: [eg: Arduino Nano 33 IoT]

mathieucarbou, 2024-07-09:
Hi,
Maybe have a look at this: https://github.com/fabianoriccardi/dimmable-light/pull/63 and you can adapt.

mathieucarbou, 2025-11-26:
@pepuma fyi my pr was closed because I removed the fork. I have since developed a far better dimming library with a superior control over fading and steps, and a correct handling of ISR.
https://github.com/mathieucarbou/MycilaDimmer