fabianoriccardi/dimmable-light #70: not working on esp32 3.0.7

JesusRosaB, 2024-12-16:
Hi I am trying to use the library in my project but it does not work.

I am using Arduino IDE with ESP32 boards version 3.0.7 and it gives me several compilation errors, which I describe below:

Dimmable_Light_for_Arduino\src\hw_timer_esp32.cpp: In function 'void timerInit(void (*)())':
Dimmable_Light_for_Arduino\src\hw_timer_esp32.cpp:32:21: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
   32 |   timer = timerBegin(TIMER_ID, 80, true);
      |           ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
In file included from \esp32\hardware\esp32\3.0.7\cores\esp32/esp32-hal.h:84,
                 from \esp32\hardware\esp32\3.0.7\cores\esp32/Arduino.h:36,
                 from Dimmable_Light_for_Arduino\src\hw_timer_esp32.h:23,
                 from Dimmable_Light_for_Arduino\src\hw_timer_esp32.cpp:22:
\esp32\hardware\esp32\3.0.7\cores\esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
Dimmable_Light_for_Arduino\src\hw_timer_esp32.cpp:36:23: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
   36 |   timerAttachInterrupt(timer, callback, false);
      |   ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
\esp32\hardware\esp32\3.0.7\cores\esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
Dimmable_Light_for_Arduino\src\hw_timer_esp32.cpp: In function 'void startTimerAndTrigger(uint32_t)':
Dimmable_Light_for_Arduino\src\hw_timer_esp32.cpp:41:3: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
   41 |   timerAlarmWrite(timer, delay, false);
      |   ^~~~~~~~~~~~~~~
      |   timerWrite
Dimmable_Light_for_Arduino\src\hw_timer_esp32.cpp:42:3: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
   42 |   timerAlarmEnable(timer);
      |   ^~~~~~~~~~~~~~~~
      |   timerAlarm
Dimmable_Light_for_Arduino\src\hw_timer_esp32.cpp: In function 'void setAlarm(uint32_t)':
Dimmable_Light_for_Arduino\src\hw_timer_esp32.cpp:47:3: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
   47 |   timerAlarmWrite(timer, delay, false);
      |   ^~~~~~~~~~~~~~~
      |   timerWrite
Dimmable_Light_for_Arduino\src\hw_timer_esp32.cpp:51:3: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
   51 |   timerAlarmEnable(timer);
      |   ^~~~~~~~~~~~~~~~
      |   timerAlarm

grpc: error while marshaling: string field contains invalid UTF-8

Compilation error: grpc: error while marshaling: string field contains invalid UTF-8

mathieucarbou, 2024-12-16:
Hello,
my PR could help you maybe: https://github.com/fabianoriccardi/dimmable-light/pull/63

zapphyre, 2025-01-06:
can we get this merged?